Ultimaker/Cura #10076: custom definitions not displayed in add p[rinter list

firepower9966, 2021-07-03:
### Application Version

4.10

### Platform

windows 10 64bit 

### Printer

E3D V6

### Reproduction steps

Custom definitions not displayed in add printer list for clean install of Cura 4.10.0 followed by copying custom definitions into resources/definitions  folder.

Custom definitions are displayed in add printer list for clean install of Cura 4.9.1 followed by copying custom definitions into resources/definitions  folder. (custom definition is E3D V6)

attached is screen shot of add printer list for Cura 4.9.1 and Cura 4.10.0 showing E3D V6 in list / not in list

[image]

### Actual results

Custom definitions not displayed in add printer list.

### Expected results

Custom definitions are displayed add printer list.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

smartavionics, 2021-07-03:
Do you need to update the setting version to 17 in that file?

firepower9966, 2021-07-03:
I don't see a setting label or value. 
Only Version which is 2
Version 2

firepower9966, 2021-07-03:
Don't see any setting label or value, 

Only at start is
   "version": 2,

smartavionics, 2021-07-03:
OK, sorry, that was a guess.

Can you please show the contents of that file?

fieldOfView, 2021-07-03:
The issue template asked for a log. In all likeliness, the requested log has information about the reason why the definition is not showing up.

firepower9966, 2021-07-03:
here is definition and cura.log file

[cura.zip](https://github.com/Ultimaker/Cura/files/6758715/cura.zip)
[e3d_v6.def.zip](https://github.com/Ultimaker/Cura/files/6758717/e3d_v6.def.zip)

fieldOfView, 2021-07-03:
From the log file:
```
2021-07-03 18:05:48,883 - ERROR - [MainThread] UM.Logger.logException [112]: Exception: Unable to deserialize metadata for container C:\Program Files\Ultimaker Cura 4.10.0\resources\definitions\e3d_v6.def.json: e3d_v6: Expecting property name enclosed in double quotes: line 63 column 21 (char 2202)
```

More importantly: Why is this definition redefining all settings in Cura? Instead it should be inheriting from fdmprinter.def.json. 
In other words: This is a badly written printer definition

fieldOfView, 2021-07-03:
Also from the definition file:
```
    "metadata":
    {
        "type": "machine",
        "author": "firepower",
        "manufacturer": "E3D V6",
        "setting_version": 16,
        "file_formats": "text/x-gcode;model/stl",
        "visible": true,
```
So it _does_ have a setting version that needs to be updated to 17.

firepower9966, 2021-07-03:
It my first attempt at writing definition, but why it work in 4.9 but not 4.10 ? 
I do what you recommend and use inherit, I think i just edited and renamed fdmprinter.def.json.

firepower9966, 2021-07-03:
"setting_version": 16,
ok i update to 17

fieldOfView, 2021-07-03:
Look at all the other definitions. See how they do it, and do it similarly. None of them make a full copy of fdmprinter.def.json.

Out of curiosity: what are you trying to accomplish? As far as I know, the E3D V6 is a hotend, not a printer. Are you making a definition for a printer that is modified with an E3D V6? In that case it is smarter to inherit from that printer definition.

firepower9966, 2021-07-04:
my original printer was a Geeetech A10M, but now total customized. have replaced the dual mixing hot end with a E3D V6 with 3 into 1 switching hotend, but currently running with single direct drive.my setup will have capability to run 3 extruders.
also i change quality setting to have 0.04 magic number multiple of layer height. 

attached is my definition file updated as you recommended

[e3d_v6.def.zip](https://github.com/Ultimaker/Cura/files/6759398/e3d_v6.def.zip)

fieldOfView, 2021-07-04:
> This can be marked closed / resolved.

I don't have that power, but you do.

> attached is my definition file updated as you recommended

This looks much better. Much easier for you to see what settings you changed too.

closed 2021-07-04 by firepower9966