Ultimaker/Cura #10077: Отсутствие Ender 6 при выборе принтера.

JerryRoppery, 2021-07-03:
### Is your feature request related to a problem?

В списке выбора принтеров отсутствует принтер Ender 6 

### Describe the solution you'd like

Добавление в список принтеров принтера Ender 6

### Describe alternatives you've considered

Пока альтернатив не рассматривал

### Affected users and/or printers

Пользователи которые используют принтер Ender 6  смогут использовать программу с актуальными настройками для своего принтера.

fieldOfView, 2021-07-03:
Please use english when submitting an issue. I understand it is an inconvenience if that is not your native language, but while english is not the native language of many people that can help you here, most of us are unable to read Russian but we do read and write English.

JerryRoppery, 2021-07-03:
please add the Creality Ender 6 printer to the program

fvrmr, 2021-07-05:
Hi @JerryRoppery we don't maintain these printer profiles. For third party printers we are depended on our community to create these. You can create them your self and create a pull request or wait till someone else create them. 
I will close this issue since this is something we won't do.

closed 2021-07-05 by fvrmr