Ultimaker/Cura #10079: Support is generated horizontally.

twinclouds, 2021-07-04:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

To generate support for a stature.

### Actual results

For the particular stature, the [image]  But for others it works fine.
[image]

### Expected results

It should be vertical.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

You will see once you open the file.  I cannot attach the project file.

fvrmr, 2021-07-05:
Hi @twinclouds could you add your project file (file - save project) as a zip here on Github? And are you adding the support with the custom support plugin?

twinclouds, 2021-07-05:
Yes.  The support plugin is installed.  The file is too big to be included.  It's link is:  https://drive.google.com/file/d/1YF8dDWixDLjYEQ06KKiiiQ7xEreGuaks/view?usp=sharing
Thank you for your help.

fvrmr, 2021-07-06:
Thank you, I think that this is happening because the model is grouped. Do you have this with other models? When I ungroup everything and add support it is working fine. [image]

But when it is grouped it adds indeed the support horizontal. In that case it is best to report this to the plugin developer: https://github.com/lokster/cura-custom-supports

closed 2021-07-20 by no-response[bot]