Ultimaker/Cura #10080: 4.10

Boheltian, 2021-07-04:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Creality Cr10 Pro V2

### Reproduction steps

Open Cura 4.10.0
Load file I want to print
Right click

or

Open Cura 4.10.0
click on settings bar

after crash Cura continues to crash on loading application 

### Actual results

Cura Freezes

### Expected results

Open right click settings or access settings bar

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

have attempted to reinstall and will work on first install until I do any of the actions mentioned.

fieldOfView, 2021-07-04:
Do you have multiple displays? If so, try to keep Cura on the primary display.

Boheltian, 2021-07-04:
> Do you have multiple displays? If so, try to keep Cura on the primary display.

I do and it crashes on the main display as well

fieldOfView, 2021-07-04:
To diagnose this issue, can you try to disable all but one display?

Boheltian, 2021-07-04:
> To diagnose this issue, can you try to disable all but one display?

It appears to work when I unplug my other 2 displays

fvrmr, 2021-07-05:
Hi @Boheltian do you have a docking station?

fieldOfView, 2021-07-05:
Could you upload a logfile as created by dxdiag (start -> run -> dxdiag -> save output)?

Boheltian, 2021-07-05:
> Could you upload a logfile as created by dxdiag (start -> run -> dxdiag -> save output)?

[DxDiag.txt](https://github.com/Ultimaker/Cura/files/6763988/DxDiag.txt)

Here ya go

Boheltian, 2021-07-05:
> Hi @Boheltian do you have a docking station?

no, I don't believe I have a docking station

konskarm, 2021-07-20:
This is in our backlog. Devs see CURA-8412.

konskarm, 2021-08-03:
This should now be fixed for 4.11 by PRs https://github.com/Ultimaker/Cura/pull/10210 and https://github.com/Ultimaker/Uranium/pull/719.

closed 2021-08-03 by konskarm