Ultimaker/Cura #10081: Adaptive Retraction - A Post Processing Script

akromatikus, 2021-07-04:
### Is your feature request related to a problem?

I currently have a post processing script I made that adjusts retraction distance based on travel distance (see my profile if interested - it's open source). It works wonderfully, however I have to view a graph of the 'retract per travel distance' in a separate python program to know what the results will be. I want to know if there is a way to incorporate/display a plot pop up (matplotlib or anything similar) within the Cura plugin instead.

### Describe the solution you'd like

The ability to display a plot/graph in a Cura post processing script. (matplotlib, etc.). Also the ability to update the plot when settings within the script are changed.

### Describe alternatives you've considered

The alternative is to use my python program, or post the plot function in something like Desmos, but having everything in the box is a lot cooler of course :)

### Affected users and/or printers

Anyone using this script with Cura. Personally I love the feature and use it for all of my prints. If it becomes popular I'd like to submit a pull request in the future, but without the plot is won't be useful for anyone but me.

fvrmr, 2021-07-05:
Hi @akromatikus it would be better to display this in a plugin instead of a post processing script. A post processing script will be added when you press send to printer or create g-code. So it has no effect on when you slice your model and show it in preview.

Ghostkeeper, 2021-07-12:
Such graphs will be specific to the kind of processing that the script does. Like, you couldn't show a plot for the Search and Replace post-processing script, or the thumbnail script. I agree that a plug-in would be more appropriate; plug-ins have the ability to make changes to the interface and execute arbitrary Python code at any moment during the application's runtime, so they can display plots like that.

What you could also do is serialise the plot in the g-code and have some application that reads that serialisation out from a g-code file.

Since we don't ship your script, not as a script nor a plug-in, I'll close this feature request. Your plug-in would have to implement it.

closed 2021-07-12 by Ghostkeeper

PaintOnBrush, 2023-02-26:
you could search g0 codes and take the pythagoream multiplied by the ooze factor. (0.0 - 1.0)