Ultimaker/Cura #10082: Hole outer wall speed

leonhma, 2021-07-04:
### Is your feature request related to a problem?

A well known and frustating one indeed. The vertical holes for screws, pegs, etc. on my print always turn out too small. After some research i found, that when the printer takes the corner (round hole), it pulls the still sticky pla string to the inside of the hole. That means the outer and inner walls of holes dont connect and the hole turns out too small.

### Describe the solution you'd like

A combination of 'Hole horizontal expansion' and 'Outer wall speed'. 'Hole outer wall speed' so that i can set it to like 20mm/s to reduce this effect. Maybe make this per default 20mm/s so new users dont have to deal with it.

### Describe alternatives you've considered

None, actually.

### Affected users and/or printers

It will affect pritty much all users, and reduce the effect of said problem.

### Additional information & file uploads

[image]
you can see the outer walls of the holes are pulled to the inside.

rlhahn, 2021-07-04:
Something like this is already in Cura. It's under experimental called Small Hole Max Size. It allows you to slow down the print by some % on holes that are a certain size or smaller. It helps a tone for really small hole. I think tho that you need to lower your print head (more squish, set a negative z-offset) since these look big enough not to need this Small Hole Max Size feature.

fvrmr, 2021-07-05:
Hi @leonhma like @rlhahn mentioned above we have `small hole max size` and `small feature speed`. 

I will close this issue since it is something that we already have, but please let us know if this works for you!

closed 2021-07-05 by fvrmr

leonhma, 2021-07-06:
Thank you very much for the suggestion. I added a z-offset of -.04, but since im printin on BuildTak, i can't really lower the print any further, without making it impossible to remove. It at least looks a bit better now.