Ultimaker/Cura #10083: Unable to Slice with current settings. Zhop speed

jamesblen, 2021-07-04:
### Application Version

4.10.0

### Platform

Windows 10 Pro 

### Printer

Monoprice mini V2

### Reproduction steps

1. Downloaded and uninstalled 4.9 completely and installed 4.10.0
2. Added the Monoprice mini V2
3. Now Cura won't slice

### Actual results

Cura does not work

### Expected results

Expected it to work as well as 4.9 but it doesn't
[Bathroom Mucket V5 L v2.zip](https://github.com/Ultimaker/Cura/files/6759884/Bathroom.Mucket.V5.L.v2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The file I sent you worked with 4.9

GregValiant, 2021-07-04:
Try going to Speed | Z Hop Speed and set it to 1.5.  If it isn't visible then use the drop down list next to the search box and set visibility to "All".
The Max Z Speed for your printer is 1.5 and the default for the Z Hop speed is higher than that.  That could be the reason it won't slice.

fvrmr, 2021-07-05:
I will discuss this with the team, keep you posted!

GregValiant, 2021-07-05:
@fvrmr I left a post on #10085 .  It appears that the problem rests in the definition file.

jamesblen, 2021-07-07:
> Closed #10083 <https://github.com/Ultimaker/Cura/issues/10083> via fe0f460

Thank you for your response. Agreed.

fvrmr, 2021-07-08:
Closing since it should be fixed for next release.

closed 2021-07-08 by fvrmr