Ultimaker/Cura #10084: Initial Layer Outer Before Inner Walls

rlhahn, 2021-07-04:
### Is your feature request related to a problem?

When printing with material that need a brim, Outer Before Inner Walls helps greatly with adhesion since since the walls touching the brim are printed first, and successive walls/bottom layers are printed touching those that came before them. ie no filament is printed on bare build plate where it can be easily removed. 

### Describe the solution you'd like

An option for the Outer Before Inner Walls to be on the first layer only. This could also be the default for first layers that use a brim as adhesion would be increased. 

### Describe alternatives you've considered

Using  Outer Before Inner Walls for the whole print but if there are large overhangs this in not ideal for the rest of the print. 

### Affected users and/or printers

Everyone who struggles to get materials like abs to stick to the print bed.

### Additional information & file uploads

This is what I'm trying to avoid for the first layer only. If there's small pieces were printed first they would likely not stick, but I would like the option to not have Outer Before Inner Walls on the rest of the print.

Avoid this:
[image]

Want this:
[image]

But only on the first layer.

fvrmr, 2021-07-05:
Hi @rlhahn thank you for your feature request. I will bring this up for discussion with the team. 
Keep you posted!

rburema, 2021-07-06:
We've not set a priority on this at all yet, and it could even still be declined, but we've tentatively opened a 'nice to have' type of internal ticket (CURA-8367) for now.