Ultimaker/Cura #10085: Z-hop speed ERROR

TdH-2021, 2021-07-04:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Tevo Taratula

### Reproduction steps

opent cura whit the setings that workt on 4.9 and now it shows Z-hop speed error.
i have try't manny diverent ways to solve it but no succes.

### Actual results

cura won't slice my project but it workt on 4.9

### Expected results

having no proplems like in 4.9

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Qi_wireless_stand_for_Iphone_X.zip](https://github.com/Ultimaker/Cura/files/6760533/Qi_wireless_stand_for_Iphone_X.zip)

gcusick, 2021-07-05:
Same problem observed using 4.10 on Macos 11.3.1

fvrmr, 2021-07-05:
Hi @TdH-2021 seems like a duplicate of https://github.com/Ultimaker/Cura/issues/10083 so I will close this one.

closed 2021-07-05 by fvrmr

GregValiant, 2021-07-05:
When you add your printer it uses the settings in the fdmprinter.def.json file.  It then alters some of those settings with your Tevo_tarantula.def.json file which was authored by "TheAssassin" and submitted by "Tevo" for inclusion in Cura.
So initially fdmprinter.def.json loads and one of the sections is "Speed_Z_Hop".  Within the "Speed_Z_Hop" section are:
          "default_value": **10**,
          "minimum_value": "0",
          **"maximum_value": "machine_max_feedrate_z"**,

Then the Tevo_tarantula.def.json file states in the "overrides".
**"machine_max_feedrate_z"**: { "default_value": **3** },
So your Max Z speed is 3 but the default "Speed_Z_Hop" is 10 and that immediately violates the "maximum_value" for the setting and so the model won't slice.

If you have "Printer Settings" installed in the Marketplace, you can change the "Printer Settings | Max Speed Z" to a higher number (greater than 10) or alternatively you can change the "Speed | Z Hop Speed" setting to 3 or less. 

The problem I see is that to this point no one has bothered to ask the printer what it's actual Max Feedrate Z is.  It is stored in M203 and is the real number that should be considered.  M203 is included in the printer's response to an M503 query.  If the printer was delivered with a Max Z Feedrate of 3, then the definition file that Tevo provided should reflect that and change the Speed_Z_Hop default to the Max Z Feedrate.

Yours is not the only printer with this problem.