Ultimaker/Cura #10087: [linux] Fresh download of Cura 4.10.0 can't sign in

TechieAndroid, 2021-07-04:
### Application Version

4.10.0

### Platform

Gentoo Linux

### Printer

Ender 3

### Reproduction steps

Download Cura 4.10.0 and try to sign in.

### Actual results

Nothing happens and the log shows: `cura.OAuth2.AuthorizationService.startAuthorizationFlow [163]: Starting new OAuth2 flow...` but it never does anything past that, no prompt, no web redirect, nothing.

### Expected results

I expect to be able to sign in.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6760729/cura.log)

fvrmr, 2021-07-05:
Hi @TechieAndroid could you try and see if you can open other links in Cura. For example: Help - Report a bug.

Spitfireap, 2021-07-10:
hi @TechieAndroid, i've got the same bug. I can't open Help - Report a bug nothing shows up on cura. On the console I've got a seg fault trying to open the bug report page.

error : `/usr/bin/xdg-open : ligne 613 : 17722 Erreur de segmentation  (core dumped)kde-open${KDE_SESSION_VERSION} "$1"`

Platform : 
Manjaro Kde, cura 4.10.0

TechieAndroid, 2021-07-10:
> Hi @TechieAndroid could you try and see if you can open other links in Cura. For example: Help - Report a bug. 

I'll try to check it when I have time. My schedule is pretty packed.

Last time I recall nothing was working like the other guy said. Probably the same issue.

I uploaded the log so that should prove helpful to debugging.

Spitfireap, 2021-07-10:
@TechieAndroid are you using KDE plasma ? Maybe it's related to the DE ?

TechieAndroid, 2021-07-10:
> @TechieAndroid are you using KDE plasma ? Maybe it's related to the DE ? 

Yeah I use the full plasma-meta and kde-apps-meta suite.

Shouldn't be a issue with that because kde is one of the top of the line DEs. Nothing else acts up on it, just Cura.

Ghostkeeper, 2021-07-12:
From Cura's side, we just call upon Qt to open up the webpage: `QDesktopServices.openUrl(url)`. Qt will call `xdg-open` on Linux to open the web page with your preferred browser. Why your `xdg-open` causes a segfault I don't know, but I don't think it is something that Cura has any control over. Cura didn't install it, since it's part of your system.

When it says "Starting new OAuth flow" in the log, the next step is that it waits for the browser to give Cura a log-in token that it can then use for API requests. So if the browser never opens, this OAuth flow just keeps waiting.

One thing that could be an issue: Where did you get Cura from? Did you download it from Ultimaker's website or Github?

TechieAndroid, 2021-07-12:
> From Cura's side, we just call upon Qt to open up the webpage: `QDesktopServices.openUrl(url)`. Qt will call `xdg-open` on Linux to open the web page with your preferred browser. Why your `xdg-open` [...]

I got the appimage from the official website.

Spitfireap, 2021-07-12:
> From Cura's side, we just call upon Qt to open up the webpage: `QDesktopServices.openUrl(url)`. Qt will call `xdg-open` on Linux to open the web page with your preferred browser. Why your `xdg-open` [...]

I tried with both github and Ultimaker website but none worked.

I've also tried simple command such as `xdg-open google.com` and it worked great... I've tried running cura from source but pynest2 is giving me headache to build ^^'. I wanted to check what auth_url was exactly in cura/OAuth2/AuthorizationService.py.

Ghostkeeper, 2021-07-13:
Here is my build of Pynest2d. Just built it for a new development environment: [pynest2d.zip](https://github.com/Ultimaker/Cura/files/6807833/pynest2d.zip) With a little luck it could just drop into your dist-packages like this. I'm using Python 3.8.

thetuxkeeper, 2021-08-24:
I tried to debug the issue and found a workaround. It looks like a the QT_PLUGIN_PATH shipped with the appimage is incompatible with the local one and it is used in a mix in my case.

The workaround is a `xdg-open` in PATH which overrides and wraps the "real" xdg-open and unsets QT_PLUGIN_PATH (`$HOME/bin` in my case with `export PATH="$HOME/bin:$PATH"` in my `.bashrc`, don't forget `chmod +x`) :
```shell
#!/bin/bash

unset QT_PLUGIN_PATH
/usr/bin/xdg-open "$@"
```

I got the hint in the backtrace - you can see a Qt library mix of local and appimage libraries:
```
Core was generated by `kde-open5 https://account.ultimaker.com/authorize?client_id=um-----------------'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f907d421263 in find_atom_pointer () from /tmp/.mount_Ultima4Reitg/usr/bin/qt/plugins/platforms/libxkbcommon-x11.so.0
[Current thread is 1 (Thread 0x7f907fad0880 (LWP 15988))]
(gdb) bt
#0  0x00007f907d421263 in find_atom_pointer () from /tmp/.mount_Ultima4Reitg/usr/bin/qt/plugins/platforms/libxkbcommon-x11.so.0
#1  0x00007f907d42148e in atom_intern () from /tmp/.mount_Ultima4Reitg/usr/bin/qt/plugins/platforms/libxkbcommon-x11.so.0
#2  0x00007f907d420ff5 in xkb_keymap_new () from /tmp/.mount_Ultima4Reitg/usr/bin/qt/plugins/platforms/libxkbcommon-x11.so.0
#3  0x00007f907d41ecaf in xkb_x11_keymap_new_from_device () from /tmp/.mount_Ultima4Reitg/usr/bin/qt/plugins/platforms/libxkbcommon-x11.so.0
#4  0x00007f907f12ac46 in ?? () from /tmp/.mount_Ultima4Reitg/usr/bin/qt/plugins/platforms/libQt5XcbQpa.so.5
#5  0x00007f907f12b00c in ?? () from /tmp/.mount_Ultima4Reitg/usr/bin/qt/plugins/platforms/libQt5XcbQpa.so.5
#6  0x00007f907f124564 in QXcbConnection::QXcbConnection(QXcbNativeInterface*, bool, unsigned int, char const*) () from /tmp/.mount_Ultima4Reitg/usr/bin/qt/plugins/platforms/libQt5XcbQpa.so.5
#7  0x00007f907f12834a in QXcbIntegration::QXcbIntegration(QStringList const&, int&, char**) () from /tmp/.mount_Ultima4Reitg/usr/bin/qt/plugins/platforms/libQt5XcbQpa.so.5
#8  0x00007f907f41c69d in ?? () from /tmp/.mount_Ultima4Reitg/usr/bin/qt/plugins/platforms/libqxcb.so
#9  0x00007f9084100f19 in QPlatformIntegrationFactory::create(QString const&, QStringList const&, int&, char**, QString const&) () from /usr/lib64/libQt5Gui.so.5
#10 0x00007f908410cf66 in QGuiApplicationPrivate::createPlatformIntegration() () from /usr/lib64/libQt5Gui.so.5
#11 0x00007f908410e4c8 in QGuiApplicationPrivate::createEventDispatcher() () from /usr/lib64/libQt5Gui.so.5
#12 0x00007f9083c81066 in QCoreApplicationPrivate::init() () from /usr/lib64/libQt5Core.so.5
#13 0x00007f90841109ae in QGuiApplicationPrivate::init() () from /usr/lib64/libQt5Gui.so.5
#14 0x00007f9084703679 in QApplicationPrivate::init() () from /usr/lib64/libQt5Widgets.so.5
#15 0x000055e8aedbd757 in main (argc=<optimized out>, argv=<optimized out>) at /usr/src/debug/kde-plasma/kde-cli-tools-5.21.5/kde-cli-tools-5.21.5/kioclient/kioclient.cpp:154
```

Spitfireap, 2021-08-24:
I forgot about this bug... Other workaround I have is simply run Wayland instead of X11. It fixes immediately the issue...

Ghostkeeper, 2021-09-02:
Good find, thetuxkeeper. We do need the `QT_PLUGIN_PATH` to be set in the execution of Cura though. Maybe this is something we could post upstream to Qt.

samsonasu, 2021-11-16:
Confirming both that this issue still exists on the 4.12 AppImage release and @thetuxkeeper's workaround still fixes it.

igor-stoppa, 2021-12-12:
Stil a problem in 4.12.1
slight improvement on @thetuxkeeper 's workaround: make a wrapper for the AppImage and only in the wrapper alter the $PATH
But yeah, the workaround on the xdg-open path stil works.

opok, 2022-01-31:
I experienced this same problem with Cura 4.13.0 installed as AppImage on Manjaro KDE

When clicking the Sign in button in Cura, it would not open any browser.
I found this related issue https://github.com/Ultimaker/Cura/issues/8094 with some tips

1. I tried `python3 -m webbrowser -t https://www.github.com` -> it opens Google Chrome just fine. I didn't try the PyQt5 thing.
2. after starting Cura by this command `env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP /path/to/Cura.AppImage` the Cura was finally able to open browser, but it openned Chromium :thinking: 

After logging in Cura is now able to create backups even when started normally from Application launcher.

Felixoid, 2022-04-29:
Hello. Just installed appimage version 4.13.1 on Arch linux plasma, and it has the same issue.

MariMakes, 2023-10-19:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, and we've introduced a lot of changes since this report. I'll be closing this issue. I hope you understand.

You can download the latest version of Cura with a bunch of improvements and a new QT version here:
[https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)

Apologies for any inconvenience caused by our delayed response.
Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-19 by MariMakes