Ultimaker/Cura #10091: Silent installation is not silent. Adafruit Industries device software requires interactive console

basictheprogram, 2021-07-05:
### Application Version

4.10

### Platform

Windows 10

### Printer

N/A

### Reproduction steps

From an Admin powershell

> Ultimaker_Cura-4.10.0-amd64.exe" /S 

### Actual results

Would you like to install this device software for Adafruit Industries dialog is displayed and requires an interactive console.

[image]

### Expected results

Silent install should be silent.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

No logs, no project file, the problem is with doing a silent installation.

basictheprogram, 2021-07-05:
I've run into a similar problem in the past, documented at https://github.com/Ultimaker/Cura/issues/1245

basictheprogram, 2021-07-05:
Just noting my security guys complaining. Signature expired 01/12/2020

[image]

fvrmr, 2021-07-06:
Hi @basictheprogram I will bring this up for discussion with the team. Keep you posted!

basictheprogram, 2021-07-06:
Work around for /me/ is discussed here: http://disq.us/p/2i28wrk and the code to insert the certificates before doing the installation can be found at https://github.com/basictheprogram/chocolatey-cura-new/tree/4.10

evtrados, 2021-07-07:
Hi @basictheprogram ,

I've tried to reproduce this issue locally on two PCs, but I couldn't. I naturally had already the Arduino related drivers installed on my systems, so I first removed them completely.
After opening a `cmd.exe` shell as an Administrator and tried to silently install the Cura v4.10 .exe installer (which I downloaded from the official Ultimaker website) I had no issues whatsoever. As a matter of fact the Arduino-related drivers were installed automatically:
[image]

basictheprogram, 2021-07-07:
Hi @evtrados,

I found that removing the certificates via the certificate manager and reboot was not enough. Like you, the next silent installation worked as expected. 

But a pristine windows install, I'm using vagrant and https://github.com/StefanScherer/packer-windows, it does still throw the alert dialog for both the AdaFruit and Arduino certificates.

I can also confirm on 2 new laptops (Lenovo) demonstrate same behavior. The alert dialog for both the AdaFruit and Arduino certificates.

I am also confused why removing the certificates does not work as expected.

I am happy to test more, using vagrant, virtualbox, and Windows10 if you tell me what else I can do.

w0, 2021-07-09:
I am experiencing the same behavior as @basictheprogram on windows computers that have not previously installed the drivers.

evtrados, 2021-07-14:
We discussed this with the team and decided that the issue is important enough to be in our backlog and eventually be fixed.
Developers, please see internal ticket CURA-8387.

basictheprogram, 2024-11-19:
For me? No. Worked around it.

closed 2024-11-19 by GregValiant