Ultimaker/Cura #10092: Extruder 2 Underextrusion at beginning of every Print

3Duke3, 2021-07-06:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ultimaker 3 ext + S5 (and potentially all dual material Ultimakers)

### Reproduction steps

Prepare sliced file as normal using combination of CPE+ and Breakaway Support material. (Unsure if bug is material specific) , then print file.

It appears the end piece of gcode to retract material 2 once a print has finished, has a retraction distance set with a much too higher value.

Then when a new print job is started, the prime amount of extruder 2 is not enough to compensate for the previous print jobs final retraction, thus no material is primed or extruded at the beginning of a print, enough to cause a print failure every time. 

There is no extrusion at prime blob, no extrusion at single line skirt, no extrusion at first prime on prime tower, no extrusion on first layer of support material. These extrusion moves all take place however the material is too far back inside the bowden to actually make it to the nozzle for extrusion.

Current solution: after each print job, manually feed material 2 through the extruder, so that the material is closer to the hot end for the next print job.

Untested on Cura version 4.10

### Actual results

When extruder 2 with breakaway support begins to print for every print job, nothing is extruded from extruder 2 until enough material is feed to the hotend.

Print starts.
-Extruder 2 attempts prime blob -> nothing extruded
-Extruder 2 attempts single skirt line -> nothing is extruded
-Extruder 2 attempts purge on prime tower -> nothing is extruded
-Extruder 2 attempts to print support or support interface - nothing is extruded until material finally reaches hotend.

### Expected results

Print starts
Extruder 2 attempts prime blob, extrudes material and is primed for the remainder of the print job.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Ext2 Underextrusion.zip](https://github.com/Ultimaker/Cura/files/6766857/Ext2.Underextrusion.zip)

fvrmr, 2021-07-06:
Hi @3Duke3 in 4.9.0 we had a bug that the second extruder wouldn't prime properly. This is fixed in 4.9.1 and also in 4.10.
See https://github.com/Ultimaker/Cura/issues/9672, I will close this issue since it is fixed and a duplicate.

closed 2021-07-06 by fvrmr

3Duke3, 2021-07-06:
@fvrmr your a superstar! Sorry for the duplicate, I did search but was looking for support material keywords.

I have only just updated to 4.10 and haven’t had the time to test it, trust you have it resolved though.

thanks