Ultimaker/Cura #10094: Not able to select a part in Cura (macOS BigSur)

chrazz, 2021-07-06:
### Application Version

from 4.8.0 and up

### Platform

MacOs Big Sur

### Printer

Ultimaker 2+, Ultimaker 3

### Reproduction steps

Add an STL.

### Actual results

Selection in the object list makes te item active. When grabbing the arrows on the model to move it around the buildplate the model deselects. Also models are not selectable directly on the model.

Happens with no clear reason. When working on a previously saved projectfile, suddenly selection stops. Or sometimes right after starting Cura. or when shifting models around the buildplate, then suddenly selection is not possible anymore.

Downgraded a few versions back to 4.8.0, and at first that seemed not to suffer from this issue, but soon I found it also did.

Cura has full accesrights in MacOs. Always worked perfect but now on BigSur it is not.
No model behaves predictive. I mean, sometimes the same model works fine dragging it around, but sudden ends that. A next session running Cura has no issues at all with the same model. Not clear and not reproducable, so no file to share.

Models are exports as STL's from (registered) Autodesk Inventor Professional 2021 build 353 release 21021.3.1

### Expected results

see above

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing to share, other than these facts

fvrmr, 2021-07-08:
Hi @chrazz do you have an external monitor? If so, it is a duplicate of https://github.com/Ultimaker/Cura/issues/9912

chrazz, 2021-07-08:
> Hi @chrazz do you have an external monitor? If so, it is a duplicate of #9912

Yes. Can confirm that. Using 2 HP z24. One via a hyperdrive dockingstation (hdmi port on usbc right side of Macbook, nearest to the screen) and one via another usbc to hdmi converter. (On right side of mbp nearest to the user)

Cura is always on the display connected to the converter, not the hyperdrive.

fvrmr, 2021-07-13:
Thank you for letting me know. Then I will close this ticket as a duplicate. You can add your information to that ticket, so we have everything on one place.

closed 2021-07-13 by fvrmr