Ultimaker/Cura #10095: 4.9.1->4.10.0 tree support is broken

dismalreload, 2021-07-06:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

TwoTrees Bluer

### Reproduction steps

1) Open model from: https://www.thingiverse.com/thing:3232756
2) resize it to 500% (uniform Scaling)
3) select Standard Quality (0.2mm) profile
4) enable generate Support
5) set Support Structure to tree
6) slice the model using different Support Patterns and check the result in Preview
[image]

### Actual results

no support is generated
if I change a few settings (loading my own profile which is based on the previously so-called "Draft profile") then support appears but it is buggy
for example in case of selecting concentric support pattern the result is this
[image]

### Expected results

tree support should be generated and should be printable
in 4.9.1 it was like:
[image]
for me it seems like that a tree support used to generate an outline, and the selected pattern only had effect on the infill.
in 4.10.0 though it is not generating a shell for the tree.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I've uploaded the project file with the original model (not scaled to 500%)
[TWOTB_support_test.zip](https://github.com/Ultimaker/Cura/files/6770940/TWOTB_support_test.zip)

Asterchades, 2021-07-07:
Check your Support Wall Line Count. For reasons I can only begin to guess, the TwoTrees base profile has this set to a fixed zero (0) instead of setting it to zero for regular supports and one (1) for Tree. _NB Ultimaker don't maintain the 3rd party profiles, so this change has come from an end user and been approved for integration._

The setting isn't visible in your screenshot so you'll have to toggle it.

Ghostkeeper, 2021-07-14:
This is not new in 4.10. Nothing changed there. You can get a similar effect with similar settings in 4.9:
[image]

What changed is the defaults for your printer, as per this pull request: https://github.com/Ultimaker/Cura/pull/9114

Those changes work okay by itself, but they make *your modifications* behave differently on their updated profiles.

The result is a combination of the Concentric infill pattern (from your own modifications) with 0 walls. The Support Wall Line Count setting lights up orange, as a warning that Cura thinks this print will probably not print well. The Concentric pattern uses concentric insets from the support's shape, starting with an offset from the outline. If it started with a line without offset, that would effectively add 1 wall whenever Concentric is used, which is not what the user expects (they "told Cura to use 2 walls, but they see 3! wtf"). So in your project file the first concentric ring is offset by 3.333mm from the original shape of the support. Since it's mostly thinner than 6.667mm, this causes the support to almost disappear.

Given the settings, this is expected behaviour. They are just not very good settings, and Cura warns the user about it properly. So I'd say this is expected behaviour.

If you'd like to change the defaults in the printer profile, we accept profile changes via pull requests. We can't test the print quality or reliability of those profiles since we don't have your printer. We'll just test if the settings load into Cura properly. If it is desirable to have 0 support walls in some cases but 1 in other cases, this can be implemented with a formula in `two_trees_base.def.json` (or a more specific definition file if it only needs to change for one of TwoTrees' printers).

closed 2021-07-14 by Ghostkeeper