Ultimaker/Cura #10096: Print area not adjusting properly

smooreace, 2021-07-06:
### Application Version

4.10

### Platform

Windows 10 Pro

### Printer

Ender 3v2 with maxed out Ender Extender mods

### Reproduction steps

After upgrading my printer I attempted to modify the printer configuration as follows --
[image]

The only way to resolve this is to creat a new custom printer profile. But when I do this I cant import my Ender 3 profile settings

### Actual results

The print area is pictured properly, but the additional print area in the Y axis is unusable
[image]
I can not use any area below the red line, or above the greyed out area on the north end of the bed
[image]

### Expected results

If I set the print area to 400x400x500 I should be able to use that area, or at very least be able to use the vase majority of it.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6772982/cura.log)

Asterchades, 2021-07-07:
The print area is being expanded correctly. The problem stems from the disallowed areas defined in the base Ender-3 profile. You'll either have to switch to a base profile which does not have those areas (such as the Ender-3 Pro profile, which is identical to the Ender-3 profile save the disallowed areas) or manually remove them from the "creality_ender3.def.json" file. _NB Ultimaker does not maintain 3rd party profiles, so a pull request from the community is required to alter anything in the base install._

GregValiant, 2021-07-07:
You have a third party printer altered by a fourth party.  Creality needs to submit a printer definition file for a printer with your modifications.
You can open "C:\Program Files\Ultimaker Cura 4.10Beta\resources\definitions\creality_ender3.def.json" in a text editor and make changes to the build plate size there.
"overrides": {
        "machine_name": { "default_value": "Creality Ender 3" },
        "machine_width": { "default_value": **235** },
        "machine_depth": { "default_value": **235** },
        "machine_height": { "default_value": **250** },
Do not change the name of the file or the printer name within the file - just change the build plate size and save the file.  It's probably a good idea to make a back-up copy before you alter the file.

smooreace, 2021-07-07:
Thanks both of you. I THOUGHT I was already running the old Ender 3 Pro profile but was mistaken. I did make the edits to the Ender 3 profile as described and it solved the issue. But ended up swapping to the 3 Pro as its a more native solution.

closed 2021-07-07 by smooreace