Ultimaker/Cura #10097: the layer visualization, is cracked. the lines is cracked and i can't see the start of lines.

aumondi, 2021-07-06:
### Application Version

4.10.0

### Platform

mac os big sur chip apple M1

### Printer

creality cr6se

### Reproduction steps

just slice a file and preveiw

### Actual results

lines apeared cracked, 

### Expected results

smooth and continuous lines

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2021-07-07:
If your computer graphics system does not support OpenGL 4.1 then Cura must work in Compatibility Mode and your graphics look like what you have.  It is possible that you could update your video driver.  My older laptop cannot support above 4.0 and so my graphics look like that.  Welcome to my world.

Ghostkeeper, 2021-07-14:
Indeed, that's the real problem.

One simple way to fix this problem is to add a cap to each line segment. Just a pointy triangle of 45 degrees does wonders to make this look better.

The problem with that though is that we then need to calculate where the tip of this cap ends up. We'd need to do that for every line, twice: Find the direction of the line segment, normalise that direction vector to some multiplication of the line width, then sum that normalised vector on the endpoint to get the coordinate of the tip of the cap.

For as many line segments as Cura creates, this takes a lot of time to calculate, especially in Python. We could calculate that in CuraEngine with C++ (like we do for the 4 corners of each line) but then that also has to be transferred to Python and then to a vertex buffer, which would effectively increase the loading time for layer view by 50%. That's just not worth it.

So we ended up calculating it in a geometry shader. This is calculated in the GPU, which is orders of magnitude faster than the CPU. And it's also pre-compiled to efficient GPU instructions, no Python slowness or GIL locking.

Only problem is that geometry shaders are part of the OpenGL 4.0 standard. This is a standard from 2010 so we'd kind of expect that most hardware will have caught up with that by now. But we all know that Apple is quite resistant to picking up standards from time to time... So we still provide the OpenGL 2.0 based shaders without geometry shaders as a back-up for old computers (or software that is too stubborn to update).

Zogar89, 2021-07-16:
Please, add a notification for compatibility mode and a link to faq page explaining the problem with OpenGL.

This will help people to knows why cura shows different in low and hi specs computers.

I was watching this on my girlfriend's computer and I never knew what was going on. Until today.

nallath, 2021-07-19:
> Please, add a notification for compatibility mode and a link to faq page explaining the problem with OpenGL.

Good suggestion. I will discuss this with the rest of the team.

closed 2023-06-24 by github-actions[bot]