Ultimaker/Cura #10098: Bad Circular Top Layer

Zogar89, 2021-07-07:
### Application Version

4.10

### Platform

Windows 10

### Printer

Artillery X1

### Reproduction steps

Top layer are broken

[image]
[1h10m35g-A.zip](https://github.com/Ultimaker/Cura/files/6774907/1h10m35g-A.zip)

### Actual results

Broken lines

### Expected results

Continuous line

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Proyect File uploaded

fvrmr, 2021-07-08:
Hi @Zogar89 I can't reproduce your issue, my lines are not broken. 

[image]

On what layer height did your problem occur?

nallath, 2021-07-09:
Problems like this are also solved with Arachne. We have essentially re-written the algorithms that handle that behavior for that version. It's still in BETA though, but it's stable enough to use.

Ghostkeeper, 2021-07-14:
Duplicate of https://github.com/Ultimaker/Cura/issues/7687.

closed 2021-07-14 by Ghostkeeper