Ultimaker/Cura #10099: [Cura Arachne] Nozzle downward motion during travel and collision with part

tokamac, 2021-07-07:
### Application Version

Arachne

### Platform

macOS Big Sur

### Printer

Prusa i3 MK3S

### Reproduction steps

I experience a very reproductible collision issue of the nozzle into my 3D printed models, that I don't know whose cause is. For every given G-code, it happens at the very same coordinates so the model is laterally pushed and ripped from the bed by the toolhead with a melted plastic dent at same location, every time. If I rotate the model in Cura by say 90° and re-export the G-code, the issue appears at another point and another height, but systematically the same coordinate for that new G-code. The weirdest thing is that the same G-code used to work on the same 3D printer (Prusa i3 MK3S)! I thought the printer firmware had been maybe corrupted somewhat, so I upgraded the FW from 3.9.1 to 3.10.0 yesterday: same issue.

Since the problem occurs at same location every time now, I stand by the printer waiting for it, and saw the collision directly occur with my eyes: the nozzle, instead of making a travel move at constant Z height (or with a positive Z hop set at 0.8mm), PLUNGES instead during that travel movement, i.e. it gets a NEGATIVE Z VALUE with respect to the already printed layers of the print over there, and it goes down vertically by several layers, I'd say 10 ou 15 so this is a very visible downward motion. As a consequence, when the toolhead wants to position itself to start a new layer over the already printed part, the nozzle collides with the part and pushes it laterally, melting it on the side and eventually ripping the model off the build plate. Then the print continues in thin air ending up with plastic spaghetti.

The same 3D model works fine exported from Cura 4.9 or 4.10 with no collision, except that for me it is not usable esthetically because Cura 4.x creates several bad seam scars on the surface no matter the settings I choose, that Cura Arachne does not produce. Cura Arachne was printing this model fine on my printer until the weird downward movement of the nozzle appears, that messes everything up now.

Since Cura Arachne still lacks toolpath preview, I imported the G-code in standalone PrusaSlicer G-code Viewer and checked the path at the problematic coordinates: there is no motion error in the G-code. The toolhead does not plunge downward in the G-code travel motion path where it does in real-life situation. I'm at a loss for an explanation.

I know that Cura Arachne is still beta software, but I'd want to know if other people experience same weird nozzle motion and part collision using it. The other thing I'd want to understand is why, although the issue is now reproductible, the same G-code exported with Cura Arachne from same computer used to work on the same printer before and the issue appeared "magically". I also checked the mechanics of the printer of course.

### Actual results

Abnormal downward motion of the nozzle during a specific travel path ended up in a collision with the 3D-printed part. 

### Expected results

I don't know what to do at the moment to get a collision-free print as before.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Video of the nozzle travel motion and collision](https://youtu.be/gwAeO3hP5sQ)
Anomalous toolhead travel motion at 0:56 and collision with the 3D-printed part at 0:57.

GregValiant, 2021-07-07:
Can you post the "bad" gcode file here?  In Cura use "File | Save Project" and post the 3mf file here too.

Ghostkeeper, 2021-07-14:
Am I correct in assuming that this happens at the end of the print?

So if it's not visible in the g-code as you say, perhaps this is something that the printer does after the end of the g-code?

Aside from indeed needing a project file, there's two things I'd like to know:
* Does it also happen with prints that are less tall? Like, a 10cm cube?
* What happens if you remove the `M84` command from the end g-code?

tokamac, 2021-07-14:
> Am I correct in assuming that this happens at the end of the print?

Incorrect. Look at the video and my explanation already provided above. Both clearly show or explain that it happens anywhere during the print (Z height and XY coordinates of the problem depending on the rotation of the model on the build plate). For each model set in space and its G-code generated, the problem happens at the very same X/Y/Z coordinate point each time for each print of this G-code, but the G-code tool path visualized in a 3rd-party viewer doesn't show any issue. It happens only when slicing the model with Cura Arachne, the issue is not there with Cura 4.9 or 4.10 slicing with exact same STL file in exact same position and rotation on the build plate.

Ghostkeeper, 2021-07-14:
Okay, disregard that then. Sorry! Just a project file would allow us to see if there's anything weird going on in that spot in the print then.

closed 2021-07-28 by no-response[bot]