Ultimaker/Cura #10100: Scale not work if is written manually

robertszczygielski, 2021-07-07:
### Application Version

4.10.0

### Platform

Ubuntu

### Printer

Prusa i3

### Reproduction steps

1. Import STL file (model must contains one dimension in double/float type - e.g.: 0.4)
2. Go to Scale (S) - left menu
3. Change dimension with double/float type manually by removing the number after dot (like: step 1:  '0.4', step 2: '0.' step 3: '0.5')
4. Go to next input box
5. Nothing happens

Works if you remove dot and replace by coma (like: step 1: '0.4', step 2: '0', step 3: '0,5')
Works by arrows

### Actual results

The model in not  scaled.

### Expected results

The model is scaled

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have used standard 20mm calibration cube and resize it.

fvrmr, 2021-07-08:
Hi @robertszczygielski I can't reproduce this on Windows. I will check with a colleague who uses Linux.

Ghostkeeper, 2021-07-14:
I can reproduce this and had reported it quite a while ago with the developers. It's not unique to Linux but depends on the system locale. Developers, see issue CURA-8191 .

closed 2024-12-09 by github-actions[bot]