Ultimaker/Cura #10101: Updating 4.9.1 -> 4.10.0 causes loss of custom settings

CheshireSwift, 2021-07-07:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Monoprice Select Mini V2

### Reproduction steps

1. Updated from 4.9.1 to 4.10.0.
2. Said yes to uninstalling previous version during upgrade.

### Actual results

Lost all custom quality settings/other configurations from 4.9.
Configurations from 4.8.x are still present.

### Expected results

Not that.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

fvrmr, 2021-07-08:
Hi @CheshireSwift when you say yes to uninstalling you are uninstalling all your configurations from 4.9. So in that case it is expected behaviour. However I do understand that this is not desired behaviour. I will bring this up for discussion with the team. Keep you posted.

Ghostkeeper, 2021-07-14:
Minor correction: Uninstalling previous versions of Cura by itself doesn't erase configurations from 4.9. But in that uninstaller it comes up with a pop-up message asking if you also want to erase any configuration for that release. Saying "yes" to that causes the configuration to be erased. It's literally asking that to the user. The problem is that the users don't read, and saying "yes" to every pop-up during (un)installation is quite common.

That makes this a duplicate of https://github.com/Ultimaker/Cura/issues/8233. Please continue the conversation there.

closed 2021-07-14 by Ghostkeeper

fvrmr, 2021-07-15:
Just an update for @CheshireSwift I have created a ticket on our backlog. Our UX designer is going to take a look at this to make it more clear. The ticket is linked in the other thread.

CheshireSwift, 2021-09-28:
Thanks, I appreciate it 🙂