Ultimaker/Cura #10102: Print via USB fails in 4.10.0, works in 4.8.0

thanks4opensource, 2021-07-07:
### Application Version

4.10.0

### Platform

Windows 7 laptop with Intel integrated graphics

### Printer

Printrbot Simple Metal Extended

### Reproduction steps

1. Open STL file
2. Slice
3. Print via USB

### Actual results

No print. Hot-end temperature stays at 25C ambient indefinitely.

### Expected results

Hot-end temperature rise to printing temperature (200C) and print start.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No differences in hardware, O.S., firmware, drivers, start G-code, or STL file between 4.8.0 and 4.10.0. Can't try with 4.9.1 because crashes on startup (known problem with integrated graphics vs graphics card?).

Firmware: Marlin 1.0
USB/COM driver: PJRC Teensy

Relevant sections of stderr.log files seem to be ...

4.8.0:
```
DEBUG - [Thread-1] USBPrinting.AutoDetectBaudJob.run [24]: Auto detect baud rate started.
...
DEBUG - [Thread-1] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM7 if baud rate 115200 works. Retry nr: 0. Wait timeout: 3
DEBUG - [Thread-1] USBPrinting.AutoDetectBaudJob.run [77]: Detected baud rate 115200 on serial COM7 on retry 0 with after 0.01 seconds.
```

4.10.0:
```
DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [24]: Auto detect baud rate started.
...
DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking COM7 if baud rate 115200 works. Retry nr: 0. Wait timeout: 3
...
DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [76]: Detected baud rate 115200 on serial COM7 on retry 0 with after 0.01 seconds.
```
Then ...

4.8.0
```
DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [710]: Slicing took 5.174295902252197 seconds
DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [711]: Number of models per buildplate: {0: 1}
DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [725]: See if there is more to slice...
DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [898]: Backend quit with return code 0. Resetting process and socket.
DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PreviewStage
DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SimulationView
DEBUG - [Thread-2] CuraEngineBackend.ProcessSlicedLayersJob.run [74]: Processing new layer for build plate 0...
DEBUG - [Thread-2] CuraEngineBackend.ProcessSlicedLayersJob.run [254]: Processing layers took 1.0310590267181396 seconds
Fatal Python error: Segmentation fault
Thread 0x0000133c (most recent call first):
  File "serial\serialwin32.py", line 290 in read
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\USBPrinting\USBPrinterOutputDevice.py", line 258 in _update
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap
```
(plus more Python crashes, but hot-end heats and printing starts).

4.10.0:
```
DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [737]: Slicing took 2.6871535778045654 seconds
DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [738]: Number of models per buildplate: {0: 1}
DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [752]: See if there is more to slice...
DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [925]: Backend quit with return code 0. Resetting process and socket.
DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [114]: 'info/send_slice_info' is turned off.
DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to MonitorStage
WARNING - [MainThread] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.
WARNING - [MainThread] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.
WARNING - [MainThread] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.
WARNING - [MainThread] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.
WARNING - [MainThread] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.
WARNING - [USBPrinterUpdate] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.
WARNING - [USBPrinterUpdate] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.
WARNING - [USBPrinterUpdate] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.
WARNING - [USBPrinterUpdate] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.
WARNING - [USBPrinterUpdate] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.
...
```

Full logs available on request.

After exiting 4.10.0 from the above, connecting a serial terminal program to the printer receives  infinite repeats of:
```
T:25.0 E:0 B:0.0
```

It's stated in https://github.com/Ultimaker/Cura/issues/9054 that:

> Every three seconds Cura (and Pronterface too) send M105 to check the temperature of the hot end.

But note that Cura (nor Pronterface nor anything else USB-connected to the printer) is running, yet the messages continue. I assume Cura has send some G-code which tells the firmware to continuously report until halted. Also note that the USB COM/LPT driver won't allow multiple applications  to connect to the printer simultaneously. On exit from 4.8.0 after a successful print no such messages appear in the terminal program. I haven't tried killing 4.8.0 during hot-end warmup and checking, but wouldn't Cura detect the premature exit, prompt for an "Abort print", and send the "End G-code" sequence anyway making such a test inconclusive?

Thanks in advance for any consideration given to this issue. I've read elsewhere that "Print via USB" isn't a high priority, but I find it very useful. Please don't make me "sneakernet" an SD card 12 inches between my computer and Printrbot in order to print. ;)  I can continue to use 4.8.0 but would like to move forward with all the improvements in future releases, and am very pleased that the integrated graphics crash was fixed between 4.9.1 and 4.10.0.

fvrmr, 2021-07-08:
Hi @thanks4opensource I can't really help you with this. First because we don't test USB printing anymore and second we don't test it on your 3D printer we only test Ultimaker printers. 
USB printing is always a bit tricky, so an SD card would be a better option. But if you don't want to move 12 inches maybe you can take a look at Octoprint. 
So I will close this issue and mark it as won't fix. Hope you understand. But please let me know if you have more questions.

closed 2021-07-08 by fvrmr

Ghostkeeper, 2021-07-14:
The only change between 4.10 and 4.8 to the USB printing plug-in was [this pull request](https://github.com/Ultimaker/Cura/pull/8463).

So I have an alternative you could try, if you're a bit technically inclined (which you seem to be based on the amount of detail in your post). You could go into `C:\Program Files\Ultimaker Cura 4.10.0\plugins\USBPrinting\AutoDetectBaudJob.py` and effectively undo that change. That means to change

```
line = serial.read_until(size = 100)
```

back into:

```
line = serial.readline()
```

If that works for you, we should probably consider reverting that change.
We don't really maintain the USB printing workflow any more, but pull requests are still accepted. As long as we don't have to do the legwork for printers that Ultimaker doesn't even sell.

thanks4opensource, 2021-07-14:
Thanks, @Ghostkeeper. Yes, I claim to be a bit technically inclined -- if for some odd reason you're interested, take a look at my repositories and form your own opinion.

I was planning to try fixing the problem myself if and when I found the time. Part of the  procrastination was due to my own confusion about what that would require. I'm running Cura on Windows for reasons which aren't important to detail here, and Windows is a complete black-box mystery to me. I had found the pull request but thought undoing it would require compiling a full Cura release from source, which I have no way of doing (on Windows).

I didn't realize that the Python files existed in source form in the Windows install (thanks for the tip). Here's the current mystery, though: I have plain/default installs of both [Ultimaker_Cura-4.8.0-amd64.exe](https://github.com/Ultimaker/Cura/releases/download/4.8.0/Ultimaker_Cura-4.8.0-amd64.exe) and [Ultimaker_Cura-4.10.0-amd64.exe](https://github.com/Ultimaker/Cura/releases/download/4.10.0/Ultimaker_Cura-4.10.0-amd64.exe) on the Windows box. Bringing their respective `plugins/USBPrinting` directories (sorry, "folders") back into Linux, I find:

```
$ cmp {Ultimaker_Cura_4.8.0,Ultimaker_Cura_4.10.0}/plugins/USBPrinting/AutoDetectBaudJob.py
Ultimaker_Cura_4.8.0/plugins/USBPrinting/AutoDetectBaudJob.py Ultimaker_Cura_4.10.0/plugins/USBPrinting/AutoDetectBaudJob.py differ: byte 36, line 1
$ diff -w {Ultimaker_Cura_4.8.0,Ultimaker_Cura_4.10.0}/plugins/USBPrinting/AutoDetectBaudJob.py
$
```
Note the `diff -w` (ignore whitespace) option above. The only difference between the two releases was a change from UNIX to DOS line endings in the Python text file. Ordinarily I'd ignore this, but reading the [PySerial documentation](https://pyserial.readthedocs.io/en/latest/pyserial_api.html) relevant to the pull request change:

```
 read_until(expected= LF, size=None)
    Parameters:	
        expected – The byte string to search for.
        size – Number of bytes to read.
    Returns:	
        Bytes read from the port.
    Return type:	
        bytes
    Read until an expected sequence is found (‘\n’ by default), the
    size is exceeded or until timeout occurs. If a timeout is set it
    may return less characters as requested. With no timeout it will
    block until the requested number of bytes is read.
```

The pull request's code obviously worked for @rrrlasse or he wouldn't have made it. My printer is running at 115200 baud so that's why I didn't have problems before the change. I'm not sure whether it's returning LF or CR characters, but if it's CR that would create the infinite hang. But the timeout, as set during construction of the Serial object, should still kick in. And again, none of this matters because the change doesn't seem to be in the 4.10.0 release.

For completeness, here's the full difference between the plugin in the two installed releases:
```
$ diff -rw Ultimaker_Cura_4.8.0 Ultimaker_Cura_4.10.0  
diff -rw Ultimaker_Cura_4.8.0/plugins/USBPrinting/USBPrinterOutputDevice.py Ultimaker_Cura_4.10.0/plugins/USBPrinting/USBPrinterOutputDevice.py
213a214,215
>         if container_stack is None:
>             return
diff -rw Ultimaker_Cura_4.8.0/plugins/USBPrinting/plugin.json Ultimaker_Cura_4.10.0/plugins/USBPrinting/plugin.json
5c5
<     "api": "7.4.0",
---
>     "api": 7,
```

So why did the [pull request #8463](https://github.com/Ultimaker/Cura/pull/8463)  not make it into the install `.exe` files? And since it didn't (or at least seems not to have), why am I seeing the difference in behavior?

I apologize for including all these details on an issue which isn't being maintained. I stand by the philosophy expressed by my username, and realize that if anybody's going to address the problem it's probably going to be me. Really just thinking out loud here, or on the vague chance someone might want to provide more input. I addition to the fact that I'm perpetually busy, and have a workable fallback with 4.8.0, the printer is currently inoperative as I'm in the process of upgrading it with an aftermarket heated bed and a new power supply which that necessitated. And a 3D printed enclosure for the power supply. It never ends. ;)  Those Ultimaker machines are beginning to look like a pretty good alternative. ;)  Anyway, if and when I get any results I'll post back here, or file a pull request, etc.

Thanks again.

rrrlasse, 2021-07-14:
@Ghostkeeper https://github.com/Ultimaker/Cura/pull/8463 is not merged into 4.10. It is only merged into master. So it can't be the cause of the problem.

@thanks4opensource Cura actually executes the script files that are in your "C:\Program Files\Ultimaker Cura 4.10.0\plugins\USBPrinting" directory. So you don't need to compile Cura from source to do experiments. Simply edit these files and restart Cura.

I also use USB alot, but I don't think I can help with that new problem :(

Ghostkeeper, 2021-07-15:
> The only difference between the two releases was a change from UNIX to DOS line endings in the Python text file.

> @Ghostkeeper #8463 is not merged into 4.10. It is only merged into master. So it can't be the cause of the problem.

Good point! I checked the diff to 4.8 unassumingly. Sorry for the false positive! There are no real source code changes between the two then, at least not in the USB printing plug-in.
It could still be that something else failed outside of the plug-in, but I find that unlikely because the API and system for output devices (which is what USB printing uses) hasn't changed in years as far as I know.

> I didn't realize that the Python files existed in source form in the Windows install (thanks for the tip).

Only for plug-ins, but in this case you were lucky :)

> I apologize for including all these details on an issue which isn't being maintained.

No need! I'm happy to provide the help I can. But I can only provide help with knowing how the system works as a whole, and narrowing down causes based on your observations and my experience. I can't provide help with deep-down debugging the connection since I don't have a printer here that accepts USB connections. I would normally leave this sort of issue open and mark it as deferred (since it's an actual problem but Ultimaker won't likely spend time on it) but that's a bit of an opinionated topic.

Another observation that could help narrow it down is to test whether it only occurs on Windows. Since you also have Linux installed, do you have the same problem on Linux?
If not, we could be looking at a problem with newlines regardless. Or perhaps even some firewall installed on Windows that blocks access to COM ports?

thanks4opensource, 2021-07-15:
Thanks: @rrrlasse for clearing up the branch/release mystery, and @Ghostkeeper for your continuing interest and suggestions.

> Only for plug-ins, but in this case you were lucky :)

I've long advocated that applications should be written mostly if not entirely in Python, with performance-critical modules implemented in C++ as necessary. Even if Cura only does this for plug-ins it's a big help.

> If not, we could be looking at a problem with newlines regardless.

Except that the newline-aware `serial.readline()` code has been ruled out since it isn't in the release 4.10.0 branch. Also, on reflection, I'm pretty sure the printer **is** sending newlines. When I connect to it with a terminal emulator app on Windows, that needs to be set to translate incoming LF to LF+CR to display the output correctly.

> Since you also have Linux installed, do you have the same problem on Linux?

The PC is pure Windows, not dual boot. But for completeness I'll lug a spare Linux box down to the workshop where the printer is and test.

> Or perhaps even some firewall installed on Windows that blocks access to COM ports?

I'm thinking not because the terminal emulator, Pronterface, and Cura 4.8.0 all work. But my experience is that Windows is black magic voodoo, so anything's possible.

> It could still be that something else failed outside of the plug-in, but I find that unlikely because the API and system for output devices (which is what USB printing uses) hasn't changed in years as far as I know.

:(  That was going to be my next step because the unchanged`AutoDetectBaudJob` derives from`Job` and I was going to check there for some subtle class interface change. Since you've ruled that out I guess I'm going to have to roll up my sleeves and dive in more deeply.

Note also that in the log excerpts above `USBPrinting.AutoDetectBaudJob.run`  is running in `[Thread-1]` in 4.8.0 but in `[JobQueueWorker [0]]` in 4.10.0. Maybe that's the next thing I should look into.

Ghostkeeper, 2021-07-21:
> But for completeness I'll lug a spare Linux box down to the workshop where the printer is and test.

Maybe just a virtual machine would be sufficient.

> Note also that in the log excerpts above `USBPrinting.AutoDetectBaudJob.run` is running in `[Thread-1]` in 4.8.0 but in `[JobQueueWorker [0]]` in 4.10.0. Maybe that's the next thing I should look into.

That's because we've given those threads a proper name since 4.9: https://github.com/Ultimaker/Uranium/commit/e1bcdedf24b703c1356041fb1e3ccc266690f096
I don't think it's relevant.

Just to be sure, you didn't happen to be running multiple instances of Cura at the same time when testing this? One instance of Cura can prevent the other from accessing COM ports.