Ultimaker/Cura #10103: Multiple extruders option for Custom Belt Style printer in Blackbelt Cura

rwlobo0303, 2021-07-07:
### Is your feature request related to a problem?

There is no option to add Multiple extruders for Custom Belt Style printer in Blackbelt Cura

### Describe the solution you'd like

Add the option to add Multiple extruders for the Custom Belt Style printer in Blackbelt Cura like it is provided in Custom FFF printer

### Describe alternatives you've considered

No way around is possible

### Affected users and/or printers

Manufacturers as well as hobbyists

nallath, 2021-07-08:
We don't create or support Blackbelt Cura. Please report feature requests / issues with them directly.

closed 2021-07-08 by nallath

fieldOfView, 2021-07-08:
Just to save you a couple of steps: as the developer of Blackbelt Cura, I can say that it is not currently getting any new features. Perhaps its functionality will return some day as a plugin for Ultimaker Cura. But there is no timeline on that.