Ultimaker/Cura #10104: 4.10 - trying to slice causes runaway memory and CPU use until process is killed, 4.8.0 fine

ccazabon, 2021-07-07:
### Application Version

4.10.0

### Platform

Linux x86_64 - Debian 10/Buster

### Printer

FLSun QQ-S Pro

### Reproduction steps

1. Open Cura 4.10.0 via the .AppImage file provided
2. Load project with 1 model in it, very simple.
3. Click Slice button.

### Actual results

The slicing process (engine) which is started starts consuming all the CPU it can, while continuously growing in allocated memory.  On my 8-core/16-thread machine, CPU usage hovers at around 800% (i.e. 8 cores fully occupied) with occasional spikes to 1200%.

Memory use grows quickly; in the log I will attach, it was just over 70GB when I cancelled the slice operation.  I have left it to go longer, to over 100GB, where the machine started swapping quite heavily, so I stopped it.

### Expected results

Quickly generate a small gcode file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Opening the same project/model in Cura 4.8.0 (which I have used for a while before now) results in the slice completing successfully in a few seconds.

It's a small calibration print.
[cura.log](https://github.com/Ultimaker/Cura/files/6780504/cura.log)

I have tried to attach the *.3mf project file, but Github rejects it as "We don't support that file type".  The supported file types listed are basically just images, videos, text and document formats.

fvrmr, 2021-07-08:
Hi @ccazabon you can share your project file as a .zip. Also, I don't see anything strange in your logs, so I my guess would be that it is model specific.

ccazabon, 2021-07-09:
[FQQS_CalibrationFLower.zip](https://github.com/Ultimaker/Cura/files/6787982/FQQS_CalibrationFLower.zip)

It's a simple calibration object, a thin "flower" for testing bed levelling on a delta printer.

As I said, I have not had this memory problem with 4.8.0 - this project works just fine with no changes on 4.8.0.

As it would appear to be the backend process that's chewing memory, are there any cmdline options for getting more verbose logging?  The app doesn't show usage with --help etc so I haven't found any.

smartavionics, 2021-07-09:
Bisected, it's https://github.com/Ultimaker/CuraEngine/commit/01e5e945aef278a1086b0296603e4f5fe0cf1d9f

ccazabon, 2021-07-11:
Thanks @smartavionics for doing that!  With luck that makes this a quick fix.

Ghostkeeper, 2021-07-15:
Indeed, that helps a lot!

closed 2021-07-19 by konskarm