Ultimaker/Cura #10107: Cura crashing when loading file due to long folder name

Aivenssar, 2021-07-08:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Anycubic Mega X

### Reproduction steps

Launch Cura, load a profile, or load an object.  Anything that brings up the browse file dialog box.  

### Actual results

Cura crashes with no displayed error.

### Expected results

The loading of either a profile or object.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Bug is due to a large windows folder name that was trying to display in the quick access portion of the open file dialog.  To resolve the issue, I removed that folder from quick access and Cura no longer crashed when trying to browse files.  The folder name I had to remove from quick access was "DLAJ6-21-11136398-MP - Supervisory IT Specialist (PLCYPLN-CUSTSPT) GS-2210-15"

fvrmr, 2021-07-09:
Hi @Aivenssar I tried to reproduce your issue but I couldn't. There are issues when you want to open a folder in Cura when Chitubox is installed. Do you maybe have Chitubox installed? If not could you share your log file. You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

spbrhs07, 2021-07-09:
@Aivenssar @fvrmr I had this same issue. Uninstalling Chitubox solved my problem

Aivenssar, 2021-07-11:
@fvrmr So I do have Chitubox installed, however once I removed the long folder name from my windows quick access, Cura started working.  I had this issue with other programs in the past, so I kind of new what to look for and apparently I have a habit of long folder names ;)  I did nothing to Chitubox.

It just seems like every time the browse file dialog box popped up, with that long folder name in the quick access it would crash.  Whether I browsed for a profile to import or a file to drop in.

[cura.log](https://github.com/Ultimaker/Cura/files/6797210/cura.log)

Aivenssar, 2021-07-11:
Realizing my log file may not have the error captured in it, I just tried to recreate by re-adding that folder to my quick access.  I cannot even recreate it now.

fvrmr, 2021-07-13:
Strange, luckily you don't have the crash anymore. 
I will leave this issue open in case someone else or you runs into this as well and we have reproduction steps to reproduce it.

Ghostkeeper, 2021-07-15:
It could also be that some Explorer extension was just doing something with that folder. Maybe it was more to do with the contents of that folder rather than the length of the name?

closed 2024-12-09 by github-actions[bot]