Ultimaker/Cura #10108: cura settings not transfering to ender 3 pro

ajr85, 2021-07-08:
### Application Version

4.10.0

### Platform

windows 10 2070 super

### Printer

ender 3 pro

### Reproduction steps

editing setting in cura to print petg. saved the file to a micro sd cad, then inserted the card in to the ender 3 pro. the proplem is the setiing for nozzle temp and fan speed dont transfer over. when i run print from card it does'nt even try to start to heat the hot end. i have to go into the the printer settings and manualy enter the setting. i have tried multiple times, with always the same results.

### Actual results

the cura settings don't transfer

### Expected results

i want the cura settings to transfer.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6787775/cura.log)

ajr85, 2021-07-09:
> Hi @ajr85 could you share your project file (file - save project) and your gcode file here?

Yeah I’m not sure the the project file is though sry.

GregValiant, 2021-07-09:
With Cura open and your model loaded, select "File | Save Project" and it will create a project file.  When you post that if you could post a problem Gcode file too that would be good.  I think you have to put them in a zip file.

Ghostkeeper, 2021-07-16:
I couldn't reproduce this from the reproduction steps alone. These are the steps I've tried:

1. Add a new Ender 3 Pro printer.
2. Hit Ctrl+K to open the material manager.
3. Duplicate the PETG material. Give it a different name to recognise that it gets activated too.
4. Change the printing temperature to 111C.
5. Change the fan speed to 99%.
6. Look for the printing temperatures and fan speed in the settings list.

The temperature and fan speed correctly get set to 111C and 99% respectively, for me. Are you seeing something else happen?

I've also tried changing the Material Type field of the material to MyPETG and to PLA, to see if the issue was caused by quality profiles overriding it. But I could not detect any such problems.

A project file would help to debug exactly why the setting doesn't exhibit for you.

closed 2021-07-26 by no-response[bot]