Ultimaker/Cura #10109: Nozzle size changing issue

cseb17, 2021-07-09:
### Application Version

4.10.0

### Platform

WINDOWS 10

### Printer

LONGER LK4 PRO

### Reproduction steps

I tried to import some profiles in RESOURCES folder but same issue

### Actual results

In 4.10.0 it seems blocked:
[image]

### Expected results

In 4.9.1, I can change the nozzle size as follows:

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fvrmr, 2021-07-09:
Hi @cseb17 in 4.9 the Longer profiles were not included. So I assume that you've add the profiles yourself in 4.9. See this pull request: https://github.com/Ultimaker/Cura/pull/9150
If you want to add the nozzles you need to add them in the `longer_base.def.json` or copy your files from 4.9 the resources in the 4.10.

Let me know if that works for you.

cseb17, 2021-07-09:
Hi @fvrmr this is the first thing I did but it doesn't work :(
I copied all files from 4.9.1 to 4.10.0 resources folder and nothing happens

fvrmr, 2021-07-09:
Could you share these files?

cseb17, 2021-07-09:
> Could you share these files?

Of course but which one ?

GregValiant, 2021-07-09:
The nozzles are in the "...\resources\variants" folder.  In my installation I see that Longer only has the 0.4 nozzle available for all models and they are in a "\variants\Longer" sub-folder.  I don't know if Cura looks there.  I think yours is "longer_base_0.4.inst.cfg"

cseb17, 2021-07-11:
> The nozzles are in the "...\resources\variants" folder. In my installation I see that Longer only has the 0.4 nozzle available for all models and they are in a "\variants\Longer" sub-folder. I don't know if Cura looks there. I think yours is "longer_base_0.4.inst.cfg"

Yes this is the first step done: copy from C:\Program Files\Ultimaker Cura 4.9.1\resources\variants\longer to C:\Program Files\Ultimaker Cura 4.10.0\resources\variants\longer...
But nothing changed !

GregValiant, 2021-07-11:
As I say, each Longer model only has the one nozzle option (the 0.4 nozzle size).  You should be able to add nozzle files. 

Open "longer_base_0.4.inst.cfg" (the nozzle file for your model printer) in a text editor.  Make the two changes shown:
[general]
name = **0.6**mm Nozzle
version = 4
definition = longer_base

[metadata]
setting_version = 17
type = variant
hardware_type = nozzle

[values]
machine_nozzle_size = **0.6**

Then do a Save As and save the file as "longer_base_**0.6**.inst.cfg".  You can make a nozzle file for whatever size you want.  If you look in the variants folder you will see how other manufacturers handle it for each model of their printers.

cseb17, 2021-07-12:
> As I say, each Longer model only has the one nozzle option (the 0.4 nozzle size). You should be able to add nozzle files.
>
> Open "longer_base_0.4.inst.cfg" (the nozzle file for your model printer) in [...]

Perfect :) It works
I juste changed and incremented the setting_version number, saved and done ;)

THANKS

fvrmr, 2021-07-12:
I will close this issue since it is solved! 

Thank you @GregValiant

closed 2021-07-12 by fvrmr