Ultimaker/Cura #10113: Cura 4.10 can not select Ender-3 Pro

tamarn707, 2021-07-10:
### Application Version

4.10.0

### Platform

Windows10 Pro

### Printer

Creality Ender-3 Pro

### Reproduction steps

1. Updated from Cura4.80 to 4.10. 

2. I uninstalled Cura
Then installed Cura 4.10 again.
But the same problem occurs

### Actual results

Launch the app and set up the printer
When Cura select Ender-3 Pro as the printer, the app crashed.

After checking, there is no problem with CR-10 and Ender-2 of the same Creality.
Only Ender-3 and Ender-3 Pro crashed. 

### Expected results

Ender-3 Pro can be set 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.zip](https://github.com/Ultimaker/Cura/files/6795060/cura.zip)

About attached Log
Select Ender-2 first.
Then select Ensder-3 Pro in Add Printer .
Then Cura crased.

fvrmr, 2021-07-12:
Hi @tamarn707 I could not reproduce your issue. Could you try to delete your configuration folder of 4.10? This will delete all the data of 4.10. You can find that folder here: `C:\Users\<your username>\AppData\Roaming\cura`

tamarn707, 2021-07-14:
Hi fvrmr.

I deleted all the files and reinstalled.
But the same problem occurs.
I installed it on another laptop and it didn't cause any problems on the laptop.
I attached 3 Log files
 1. cura_Ender3Pro.log
        Select Ender3Pro and LOG when problems occur
 2. cura_Ender2.log
       Select Ender2 on the same PC and no problem LOG
 3. cura_NotePC.log
      Select Ender3 Pro on another laptop and  no problems with LOG

Looking at cura_Ender3Pro.log, 
I selected Ender3Pro on line 451 and CRITICAL occurred after line 461.
After that, ERROR occurred and Cura crashes.
[cura_log.zip](https://github.com/Ultimaker/Cura/files/6813861/cura_log.zip)

fvrmr, 2021-07-15:
Thank you I linked an other issues since the error is the same. It looks like it can't access a temporary folder. Will bring this up for discussion with the team. Keep you posted.

Ghostkeeper, 2021-07-21:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9644. Please continue the conversation there.

closed 2021-07-21 by Ghostkeeper