Ultimaker/Cura #10114: Support fails to be generated correctly at (0;0;0) coordinates of the model

HidalgoS2207, 2021-07-10:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Ender 3 PRO

### Reproduction steps

1 - Slice Model with certains supports at coordinates (0;0)

### Actual results

The support its void in Z axis until some layer in coordinates of the model placed in (0;0) 

if i move the model to lets say (5;5) the support is generated OK 

### Expected results

Support consistent and equal in any position of the plate

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fvrmr, 2021-07-12:
Hi @HidalgoS2207 could you share your project file (file - save project)?

Ghostkeeper, 2021-07-16:
From the picture, it looks like the same problem as https://github.com/Ultimaker/Cura/issues/10035.

closed 2021-07-26 by no-response[bot]