Ultimaker/Cura #10115: Z-hop while traveling

CoryC3K, 2021-07-10:
### Is your feature request related to a problem?

When doing a travel move, with no retraction, there is no Z motion (even when Z-hop is enabled). This sometimes leads to dragging the nozzle across already-printed areas, leading to lines.

### Describe the solution you'd like

I'd like a checkbox to "Z hop when travelling". This would ensure that every time there was a travel move, there was also a Z+ move to ensure against dragging.

I understand that with cartesian printers, Z motion is expensive (time & code wise). However, there are lots of delta owners like myself who don't have the issue of Z movements being costly.

### Describe alternatives you've considered

I've tried increasing the retractions and clicking all the Zhop checkboxes. However, during travel moves on tops and bottoms (especially with multiple skin layers), there is no accompanying Z-motion when travelling without a retraction. 

### Affected users and/or printers

Mainly delta users, or cartesian users who want higher quality on the inside floors of things.

fvrmr, 2021-07-12:
Hi @CoryC3K thank you for your feature request. I will bring this up for discussion with the team. Keep you posted!

fvrmr, 2021-07-15:
Can you explain more why you would like to have this feature? Because a z-hop is while traveling.

Zogar89, 2021-07-16:
I think they have combing active. With that combing travel moves no retract on travel. But if you set combing only on infill this should not be a problem.

Ghostkeeper, 2021-07-20:
Disabling combing (set to "off") will always cause it to retract. If "Z Hop When Retracting" is enabled, this will then also cause it to use a Z hop, unless Z Hop Only Over Printed Parts is enabled and there is no previously-printed part in between.

closed 2021-07-29 by no-response[bot]