Ultimaker/Cura #10117: Does not find printer

nesergen, 2021-07-10:
### Application Version

4,10,0

### Platform

linux mint 20

### Printer

board skr-1.4-Turbo, marlin 2.0

### Reproduction steps

1. open Cura
2. then power On printer

### Actual results

If the Kura program is open before turning on the printer, the program will not see it. Although the computer printer recognizes.

If I first turn on the printer, and then run the program, the printer is recognized without problems and I print.

### Expected results

I am waiting that the program will automatically connect the appeared printer.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No information.

fvrmr, 2021-07-12:
Hi @nesergen how do you connect your printer? Via wifi or usb?

nesergen, 2021-07-12:
> Hi @nesergen how do you connect your printer? Via wifi or usb? 

USB-port

GregValiant, 2021-07-14:
Unfortunately, that's the way it works.  The USB Printing plugin only runs it's script at Cura startup.  There is no button or command that might search for ports at a later time in the Cura session.  So it isn't a bug, but you could put it in as a feature request.
Since no Ultimaker printers currently use USB Printing they have no interest in putting time ($$$) into making changes to the plugin for third party hardware.  I'm pretty sure that any changes would have to come from the community.

fvrmr, 2021-07-15:
@GregValiant is indeed correct. A feature request will indeed be not developed by Ultimaker, since our printers don't use USB printing. I will close this issue and mark it as won't fix. Hope you understand.

closed 2021-07-15 by fvrmr

Ghostkeeper, 2021-07-16:
Actually, expected behaviour is that Cura checks for USB-connected devices every 5 seconds:

https://github.com/Ultimaker/Cura/blob/f9557295fa3fa329260856ed790af86668274b11/plugins/USBPrinting/USBPrinterOutputDeviceManager.py#L76

Last I checked, that still works with the Ultimaker 2. It's been very long since I tested that with an Ultimaker Original (or any other printer) though. But I could always just switch the printer on and Cura would see it without restart.

But yeah, it's unlikely that Ultimaker will develop a fix for this. Both because it's impossible to have every printer on the market to test this stuff with (since it only happens on your printer as far as we know) and because we don't have any printers with USB printing on the market any more.

GregValiant, 2021-07-16:
OK.  Now I know.  I never waited long enough.  It takes more like 30 seconds to come up but it did find my Ender.
(Patience, I gotta show more patience.)

Ghostkeeper, 2021-07-20:
Yeah once it's found the new USB device it also tests a bunch of baud rates. Depending on whether your printer uses a common baud rate or not, that can take a while too.