Ultimaker/Cura #10118: Cura does not let me see the "layer thickness" option or the "simulation" after laminating.

J-VEGA-ZORRO, 2021-07-11:
### Application Version

4.10.0

### Platform

Windows 7 64

### Printer

Anet A8 Plus

### Reproduction steps

[image]
[image]

I try to visualize the layer thickness to control the result before printing but Cura does not show it to me.

### Actual results

I activate and deactivate the option to force compatibility mode but to no avail.

### Expected results

I intend to see the variable layer thickness and activate the simulation.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It´s all.

fvrmr, 2021-07-12:
Hi @J-VEGA-ZORRO I assume you did a restart of Cura when you deactivate the compatibility mode?
Could you share your log file? You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

J-VEGA-ZORRO, 2021-07-12:
[cura.log](https://github.com/Ultimaker/Cura/files/6801169/cura.log)

fvrmr, 2021-07-16:
In the logs I see that you are using OpenGL version 3.3: 
```
2021-07-08 02:28:40,886 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2021-07-08 02:28:40,891 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  3.3.11672 Compatibility Profile Context
2021-07-08 02:28:40,896 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   ATI Technologies Inc.
2021-07-08 02:28:40,900 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: ATI Radeon HD 3600 Series
2021-07-08 02:28:40,905 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    3.30.0
```

Cura only works with OpenGL 4.1 so that's why your preview is in compatibility mode. You could try to update your video driver. 
But there's is nothing on our side we can do for you, so I will close this issue. Hope you understand.

closed 2021-07-16 by fvrmr