Ultimaker/Cura #10119: Unable to slice due to Z-Hop Settings

gothickitty93, 2021-07-11:
### Application Version

4.10.0

### Platform

macOS 11.4 (Mac Mini 2018)

### Printer

MP Select Mini v2

### Reproduction steps

1. Import stl file into the app.
2. click slice

### Actual results

I get a notification that the z-hop speed setting is incorrect, and the model does not slice.

### Expected results

I expected the file to slice, just as it did in 4.9.1

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

In Cura 4.9.1, I cannot even view the Z-Hop speed. Even after I enable the setting to be visible, it doesn't show up. Now, I don't know if the actual default setting of 10 in Cura 4.10 is correct, or if it should be the accepted value of 1
[BugReport.zip](https://github.com/Ultimaker/Cura/files/6797676/BugReport.zip)

GregValiant, 2021-07-12:
It isn't a bug.  The problem is in your printer definition file (that was supplied by MP).  It should have an override for the "Speed_Z_Hop" variable but it doesn't.  The default Z Hop Speed gets set to 10 and the maximum gets set to "machine_max_feedrate_z" then your initial definition file (malyan_m200.def.json) declares machine_max_feedrate_z = 1.5 so right away the default setting is illegal.
If you go into the MarketPlace and load the "Printer Settings" plugin - one of the settings is "maximum speed z".  If you change that to "10" you should be good to go.
Your printer isn't the only one with this problem. #10083 and #10085 are duplicates of this.

gothickitty93, 2021-07-12:
Thanks! Apparently I don't know how to search. :)

closed 2021-07-12 by gothickitty93

gothickitty93, 2021-07-12:
Here is my temporary fix until release: You can download https://github.com/Ultimaker/Cura/commit/fe0f4605476f19482a350429af6a807473ec6c5a, and just replace the existing one in your application's folder.

DeCaff24, 2021-08-13:
All previous versions of Cura did not have this issue, so why do we have this problem since the 4.10.0 release? seems to me the problem was not there yet and now with a new version of Cura there is a bug.. bug is within the Cura software. that's my logic opinion

GregValiant, 2021-08-13:
_"All previous versions of Cura did not have this issue, so why do we have this problem since the 4.10.0 release?"_
This is from Malyan200 definition file
**"inherits": "fdmprinter",**
    "metadata": {
        **"author": "Brian Corbino, Tyler Gibson"**,
...
"overrides": {
...
**"machine_max_feedrate_z": { "default_value": 1.5 }**,
This is from the Monoprice Mini definition file:
**"inherits": "malyan_m200",**
    "metadata": {
        "author": **"Tyler Gibson",**
So those are the people you need to talk to.  When they provided their new definition file to Ultimaker it was missing this line that WAS in previous versions:
**_"speed_z_hop": { "value": 1.5},_**
The problem is in the definition file supplied by Monoprice.

nallath, 2021-08-16:
The bug was in the older versions of Cura. They should have given an error, but didn't.

GregValiant, 2021-08-16:
The version of Malyan200.def.json supplied with Cura 4.10 appears to revert to the version supplied prior to Cura 4.9.x.  The line that sets speed_z-hop to 1.5 is missing in the 4.10 version (and in Cura releases previous to 4.9.x).  As a consequence, in 4.10 the default "Z-Hop Speed" of 10 in the settings dialog (inherited from fdmprinter.def.json) is illegal as it is over the max Z feedrate of 1.5 and so Cura won't slice.
In the version of Malyan200.def.json installed with Cura 4.9.x "speed_Z_hop" was set to 1.5 in the definition file and so there was no problem.

So my take on it (for what it's worth) is that there was a problem with the Malyan200 definition file through Cura 4.8, it was fixed for 4.9, and now with 4.10 it's broken again.