Ultimaker/Cura #10120: Fails to slice in 4.10, fine in 4.8

hr2burn, 2021-07-12:
### Application Version

4.10

### Platform

Windows 10 64 bit

### Printer

Ender 5

### Reproduction steps

1) pulled config file...no clue what it's taking about (WTH is pickle anyway??)
2) searched for same issue here, others didn't have models they could print in earlier versions, but not 4.10 (mine seems to be a 4.10 issues, noting else (computer, printer, model etc...)

### Actual results

Starts to slice then shows microsoft window "cure slicing engine has stopped working". I have SOME very simple calibration models that do slice..
[cura.log](https://github.com/Ultimaker/Cura/files/6797943/cura.log)
[image]

### Expected results

File should slice and be ready to send to printer.
I cannot attach the print file as it is for someone who doesn't want it shared.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I am hoping log file has enough info, if not I will watch for other with the same issues (works in older version but not latest version.)

fvrmr, 2021-07-12:
Hi @hr2burn does it only happen with specific models?

hr2burn, 2021-07-12:
Yes, I found if I remove the .14 horizontal hole expansion, it then will slice.

fvrmr, 2021-07-16:
Strange, without any project file I can't really look into this. 
So I will leave this issue open and if you (or someone else) run into this please share a project file so we can try to reproduce this.

Ghostkeeper, 2021-07-19:
> (WTH is pickle anyway??)

[Pickle](https://docs.python.org/3/library/pickle.html) is a Python library that stores logical objects on disk. We use it in Cura to speed up the boot time, by storing the pre-calculated state of the printers you added to disk.

To reproduce this issue, we'll need to know your reproduction steps. The steps you've written down are steps that you took *after* encountering the error. We need to know the steps you took *leading up to* the error. That way we can try to repeat those steps to hopefully get the same error. A project file bundles all of those steps (at least relating to the slice result) in a single file, so that is handy. But indeed, if you can't share that file we won't know how to reproduce the issue, and we can't fix this.

hr2burn, 2021-07-19:
Wish I could share the file, but I can’t. I’m using 4.8 so I have the ability to run the file.

closed 2024-12-09 by github-actions[bot]