Ultimaker/Cura #10125: Thanks for wasting my filament and print time

JoshLo1995, 2021-07-12:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Dropped model into Ultimaker Cura
2. Hit Slice
3. Cura randomly removes material from my model?? 

### Actual results

Cura produces garbage GCODE where it removed a significant portion of my model for literally no reason. 

### Expected results

Model should be sliced properly without randomly removing material

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I tried to upload my project file here, but if tools aren't going to be provided for my to upload it without additional work -- then I can't be assed. If you're going to crowdsource bug reporting you might as well include resources to use to file share.

nallath, 2021-07-13:
I've copied a relevant part from the template:  `Please make sure to .zip your project file. For big files you may need to use WeTransfer or similar file sharing sites`.

Most cases where Cura doesn't slice a part of the model is due to the model not being manifold. Cura should give a warning when this is the case. When you upload the model we can have a look.

As a final bit of advice, if you want help from a product that you are using free of charge (or at all), it generally works best to ask nicely.

GregValiant, 2021-07-13:
The file types that can be uploaded are determined by GitHub.  The Cura people have nothing to do with it.
Virtually 100% of your type of problem is found to be within the model (though it is sometimes affected by the settings in Cura).  Without a project file it's impossible to tell what the problem might be.  In Cura use "File | Save Project" and save the file as a 3mf.  In Windows Explorer right click on the file and select "Send To" and then "Compressed (zip) folder".  Post the zip file here by dragging and dropping it into your response box.  I've never found it to be a particularly onerous, difficult, nor time consuming task.  I'm sure that if is a problem with Cura that those people want to investigate it so they can fix it.

closed 2021-07-27 by no-response[bot]