Ultimaker/Cura #10126: Wrong number of layers for spherical shapes

mulder999, 2021-07-12:
### Application Version

4.10.0

### Platform

Windows 10 - 64Bit - 20H2

### Printer

AnyCubic Mega i3 S

### Reproduction steps

Attached model has a height of 23.456mm according to Cura.
With constant layer height of 0.2mm that should gives 117.28.

[3D_test_V3_0.4mm-WrongLayerCount.3mf.zip](https://github.com/Ultimaker/Cura/files/6804171/3D_test_V3_0.4mm-WrongLayerCount.3mf.zip)

### Actual results

Slicing produces 115 x 0.2mm layers

[image]

### Expected results

Slicing should produces 117 x 0.2mm layers as competitive slicers (eg: IdeaMaker).

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Cura project file attached

fvrmr, 2021-07-13:
Hi @mulder999 if you enable thin walls it shows 117 layers. I assume that your top is to thin to print. 
Let me know if that works for you.

mulder999, 2021-07-13:
Hi @fvrmr, you are absolutely correct, my very bad I have usually this feature always on and I forgot to check it. Ticket can be closed.

closed 2021-07-13 by mulder999