Ultimaker/Cura #10127: Cura 4.10 Not responding after being opened awhile

radit204, 2021-07-13:
### Application Version

4.10

### Platform

Windows 

### Printer

Hypercube

### Reproduction steps

1. Opened cura normally on a freshly rebooted computer
2. Use it regularly, to slice models, save gcode to my local drive.
3. Minimize or let cura go idle in the background as i do other things for a couple of hours
4. When i go back to cura and need to slice another model, cura suddenly becomes laggy, always showing "Not Responding", importing a simple model takes a good 10 minutes, and a drag to reorient the camera takes 1 minute to respond, it freezes up to the point where i need to close it from task manager
5. Re opens Cura, but encountered the same problem, Not Responding
6. Rebooted computer, opened cura and its all normal again. 

### Actual results

Cura not responding and freezes 

### Expected results

Cura working normally 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6805335/cura.log)

I have USB Printing disabled in cura
I also have a couple of plugins; Arc Welder, Settings Guide, SidebarGUI

radit204, 2021-07-14:
> Hi @radit204 when this happens does your computer also slow down? Or is it just Cura?

Only cura slows down, the rest of my computer is all fine

Ghostkeeper, 2021-07-20:
What's especially strange is that the issue remains after restarting Cura but not restarting your computer. Cura has had memory leaks in the past (especially with printer discovery on the LAN, and with the video feed from the printer's camera) but if there is a memory leak, restarting Cura still releases all that memory. The operating system should guarantee that.

There must be something that runs in the background *and keeps running even after Cura has been closed down*, and somehow interacts with Cura to slow it down. What could it be? Perhaps a firewall system? Maybe even video drivers?

nallath, 2021-07-20:
> There must be something that runs in the background _and keeps running even after Cura has been closed down_, and somehow interacts with Cura to slow it down. But which resets once the computer is restarted. What could it be? Perhaps a firewall system? Maybe even video drivers?

Could be some magic that Windows does for zeroconf? I vaguely remember that windows did some strange things in that area.

fieldOfView, 2021-07-20:
> Could be some magic that Windows does for zeroconf?

That can be easily tested: Go to the "Installed" tab of the Marketplace and disable the "UM3 Network Printing" plugin. If you have the OctoPrint Connection plugin, you may also want to disable that, or unset the option to "Automatically discover local OctoPrint instances" in the Preferences -> Printers -> Connect to OctoPrint dialog.

closed 2022-01-03 by radit204