Ultimaker/Cura #10132: "Slice" Crashes Cura when right click menu is open

jergling, 2021-07-14:
### Application Version

4.10.1

### Platform

Windows 10, i7-6k, GTX 970

### Printer

FLSUN QQ-s

### Reproduction steps

1. Load a model in Prepare or Preview mode
2. Select model
3. Right click anywhere in the viewport
4. Hit the "slice" button at the bottom right

### Actual results

Program hangs and shuts down immediately upon clicking anywhere else in the window. Only happens if the right-click context menu is open

### Expected results

Right-click menu closes, slice proceeds as normal

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6812752/cura.log)

fvrmr, 2021-07-15:
Hi @jergling I can reproduce this as well. I will discuss this with the team. Keep you posted.

fvrmr, 2021-07-16:
I have discussed it with the team and created a ticket on our backlog for this. 
Devs see CURA-8396