Ultimaker/Cura #10133: Arrange All Models Crashes on Apple M1

kobaltz, 2021-07-14:
### Application Version

4.10

### Platform

Apple macOS

### Printer

Creality CR-10S

### Reproduction steps

When importing multiple models, they do not auto arrange. So, I select them and try to auto arrange them. 

### Actual results

Arrange All Models does not appear to do anything

### Expected results

Arrange All Models to arrange the models, or at least display some kind of error.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

```
2021-07-13 23:19:34,521 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [112]: Exception: Unable to arrange the objects on the buildplate. The arrange algorithm has crashed.
2021-07-13 23:19:34,524 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]: Traceback (most recent call last):
2021-07-13 23:19:34,527 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.10/build/inst/lib/python3.8/site-packages/cura/Arranging/ArrangeObjectsJob.py", line 33, in run
2021-07-13 23:19:34,531 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.10/build/inst/lib/python3.8/site-packages/cura/Arranging/Nest2DArrange.py", line 126, in arrange
2021-07-13 23:19:34,534 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.10/build/inst/lib/python3.8/site-packages/cura/Arranging/Nest2DArrange.py", line 77, in findNodePlacement
2021-07-13 23:19:34,536 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.10/build/inst/lib/python3.8/site-packages/UM/Math/Polygon.py", line 179, in intersectionConvexHulls
2021-07-13 23:19:34,538 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/4.10/install/lib/python3.8/site-packages/shapely/geometry/base.py", line 676, in intersection
2021-07-13 23:19:34,541 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/4.10/install/lib/python3.8/site-packages/shapely/topology.py", line 64, in __call__
2021-07-13 23:19:34,543 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/4.10/install/lib/python3.8/site-packages/shapely/topology.py", line 18, in _validate
2021-07-13 23:19:34,545 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]: ValueError: Null geometry supports no operations
2021-07-13 23:19:34,554 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [421]: file:///Applications/Ultimaker Cura.app/Contents/Resources/qml/UM/MessageStack.qml:75: TypeError: Cannot read property 'horizontalCenter' of null
```

fvrmr, 2021-07-15:
Hi @kobaltz does this happen with all models? Or just specific models?

kobaltz, 2021-07-15:
> Hi @kobaltz does this happen with all models? Or just specific models?

It happens on every model that I've tried downloading from Thingiverse. This is an example of one where there were several STL files to be placed on the bed and they all stacked together and didn't space out.

https://www.thingiverse.com/thing:470700

fvrmr, 2021-07-16:
I will bring it up for discussion with the team. Keep you posted!

Ghostkeeper, 2021-07-21:
This has been added to our Apple M1 support epic. Developers, see CURA-8415.

kobaltz, 2021-09-16:
Just to give a bit more information. On the M1, when using the Creality Slicer (CURA 4.8.2) this issue does not exist, so somewhere between 4.8.2 and 4.11.0 it seems like the issue was introduced. Thanks for your help!

closed 2024-12-09 by github-actions[bot]