Ultimaker/Cura #10134: Crash

Mejituu, 2021-07-14:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

CoreXY 兔兔

### Reproduction steps

Open Cura and load 123.STL Cura crashes and crashes during the loading process.

### Actual results

Open Cura and load 123.STL Cura crashes and crashes during the loading process.

### Expected results

Open Cura and load 123.STL Cura crashes and crashes during the loading process.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/6813355/stderr.log)

fvrmr, 2021-07-15:
Hi @Mejituu it seems like it can't access a temporary folder. This issue: https://github.com/Ultimaker/Cura/issues/10113 is also having the same error but with loading printers. 
I will bring this up for discussion with the team. Keep you posted.

Ghostkeeper, 2021-07-21:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9644. Please continue the conversation there.

closed 2021-07-21 by Ghostkeeper