Ultimaker/Cura #10136: Profile handling improvement

rubydesign, 2021-07-14:
### Is your feature request related to a problem?

No really. The only problem is keeping an overview of changes made to profiles. In general i agree with #10000 

### Describe the solution you'd like

Next to the profile dropdown there could be an indication of whether the current profile has been modified.
I would suggest a save button that is either greyed or green. Hovering the button would give an overview of the changes.

PS: i always work with "all" settings so there can be changes burried deep

### Describe alternatives you've considered

Not really.
But the usuability of the dropdown having the save is questionable. 
But mainly there the greying out has been implemented, so the feature should be "easy"

### Affected users and/or printers

all and everyone

nallath, 2021-07-14:
It could be that I misunderstood what you are refering to, but there is an indicator to show if you have modifications to your profile:

[image]

The star on the right indicates that there are changes that are not yet part of a profile.

rubydesign, 2021-07-15:
Ok, that one went past me, never noticed.

So that is a third of the way there to what i am suggesting: Now clicking on it should save the changes, and the hover should specifically mention the changes. (or really a button is the accepted ui element, possibly next to the "Profile", or on the right, not in the dropdown)

The current behaviour of opening that profile is (from this users perspective) slightly random. As that dialog manages profiles, not their content. And i didn't want to throw too much in here, but that dialog is also strange. The export/import renaming is good, but the info displayed on the right is very very basic and not at all enough to specify differences in profile. And having a button there to update without any indication of what the update/discard would do (specifically, ie which settings would be affected) is not according to ui guidelines (transparency). But i understand that this is a different issue, i just wanted to make the point that dialog in no way is what i am suggesting.

Ghostkeeper, 2021-07-21:
> Ok, that one went past me, never noticed.

That, by itself, is also an issue. Perhaps it's not clear enough.

I can see how it would maybe become more clear if we change that icon to a floppy disk, make clicking it equivalent to saving the current changes, and add a tooltip to say so. That would help a bit, I think?

You still won't get an overview of what the profile overrides are, on the main window. For that you still have to go into the profile manager. I don't see a way to neatly implement that without overloading the user with information in the main window.

Let's keep the discussion about the layout of the profile manager in a separate issue.

rubydesign, 2021-07-21:
Even just changing the star to a floppy would be a start, yes.
Though in terms of ui guidelines, the floppy should not be in the dropdown. But ok.

I think you should give users more credit. Surely many are tinkerers and by now quite advanced. 
And as good practise i (one?) never change more than a handful of settings at a time to see what the effect is.
Si i don't think the list will be overwhelming, no. It would just be very neccessary info when working with profiles.

And while i agree that the (broken) profile view is another issue. Where is that issue? (But ok, that iis not the issue i am raising or am interested in)

rubydesign, 2021-07-21:
btw, the list of changes was only supposed to be a pop-up. I hope that was clear in my description.