Ultimaker/Cura #10137: Reload function only reloads the last changed object

Harryhar, 2021-07-14:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

Load 2 or more objects
Change 2 of the objects loaded in Cura
Cura asks to reload the object that has changed.

### Actual results

Cura only loads the last changed object, ignoring the first changed object.

### Expected results

Cura should load all changed objects

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional info

fvrmr, 2021-07-15:
Hi @Harryhar I can't reproduce your issue. Do you have any model files that can does reproduce this.

Harryhar, 2021-07-15:
Hi @fvrmr 

Copy the 2 objects from the directory "initial" to a place on your harddrive.
Import the 2 objects from your harddrive into Cura.
Copy the 2 files from the directroy "replace" to the same place on your harddrive where you copied the files from "initial"

After replacing Cura responds with :

File has been modified
Would you like to reload Object 1.stl?

Only the 1st object is reloaded, the changes to the second object are not reloaded.

[cura_replace_problem.zip](https://github.com/Ultimaker/Cura/files/6822864/cura_replace_problem.zip)

fvrmr, 2021-07-16:
Thank you, I can reproduce your issue. But if you press F5 or right click and press `reload all models` the other model will be reloaded. I will bring this up for discussion with the team. Keep you posted!

fvrmr, 2021-07-21:
We have created a ticket on our backlog for this. Developers see CURA-8419

MariMakes, 2024-11-19:
👋 Quick update on our side.
This is resolved in the 5.9 release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.9.0](https://github.com/Ultimaker/Cura/releases/tag/5.9.0)

I'll close this issue since it's resolved.
Thanks again, and please let us know if you run into any other issues 💪

closed 2024-11-19 by MariMakes