Ultimaker/Cura #10138: no retraction in 4.10

brickhouse1978, 2021-07-14:
### Application Version

4.10

### Platform

windows 10

### Printer

creality ender 3 pro

### Reproduction steps

just tried to print the same way i always do and it wont retract

### Actual results

does not retract

### Expected results

the machine should retract

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fvrmr, 2021-07-15:
Hi @brickhouse1978 can you share a project file (file - save project) where your issue happens. I don't know what settings you use to print, so I can't reproduce your issue.

brickhouse1978, 2021-07-21:
It's a software issue.   I reloaded 4.9 and it works just fine

Ghostkeeper, 2021-07-21:
Nice that you have a workaround. Could you still provide a project file though that reproduces the issue?

I tried your current reproduction steps: "Print the same way I always do" but it retracts fine. It seems to work correctly from what I can see.

closed 2021-08-04 by no-response[bot]