Ultimaker/Cura #10139: 4.10.0 not seeing configs

linuxgurugamer, 2021-07-14:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

none

### Reproduction steps

Have 4.9.1 installed, with 3 printers and 3 materials
Installed 4.10.0 
Checked C:\Users\jbb\AppData\Roaming\cura\4.10
was able to see same printers and files

[4.10.zip](https://github.com/Ultimaker/Cura/files/6818991/4.10.zip)

### Actual results

Cura 4.10 wanted to do a Add a printer dialog at startup, didn't seem to see the existing printers

### Expected results

4.10.0 should see existing printers and materials

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[4.10.zip](https://github.com/Ultimaker/Cura/files/6818995/4.10.zip)

linuxgurugamer, 2021-07-14:
I tried creating a new printer, and then it was able to see the printers, but it thought the printers and materials were corrupted
[cura.log](https://github.com/Ultimaker/Cura/files/6819008/cura.log)

rburema, 2021-07-15:
It seems that it's missing the `creality_ender6` definition. Indeed, there is nothing in the definitions folder.

Does anything like that exits in your 4.9.x config folder?

hipersferus, 2021-07-16:
Similar case, but this time Cura 4.10.0 fresh install on Win 10 Pro, does not see the new added printer def files. Did the same setup on another win 10 pro PC, same def files worked for Cura 4.9 and Cura 4.10, the new Flsun Super Racer printer was found in FLSUN section, along with the other Flsun printers. 

Attaching logs and config files for FLSUN Super Racer delta printer:
[cura.log](https://github.com/Ultimaker/Cura/files/6832673/cura.log)
[SR-20210522T051935Z-001.zip](https://github.com/Ultimaker/Cura/files/6832676/SR-20210522T051935Z-001.zip)

linuxgurugamer, 2021-07-16:
> It seems that it's missing the `creality_ender6` definition. Indeed, there is nothing in the definitions folder.
> 
> Does anything like that exits in your 4.9.x config folder?

No, missing there as well.
I think I'm just going to do a clean 4.10 install, I see from the roaming directory that I have info for 3.0, 4.2, 4.6, 4.9 and 4.10, I'm guessing that something messed up during the upgrades

rburema, 2021-07-20:
@linuxgurugamer Yes, the process assumes that the upgrade to 4.9 has succeeded, so if there's nothing there, then it won't take it along. Curious why it _does_ work on 4.9 then.

@hipersferus Sorry, but I'm not sure what you mean to say. Did your configuration get corrupted as well, or did you not find a built-in printer profile on a fresh install?

Ghostkeeper, 2021-07-21:
Perhaps you modded the Creality Ender 6 into the 4.9 installation, and forget to also mod that into the 4.10 installation? Anything in the installation folder (Program Files) doesn't get copied over to the new installation. It only copies the configuration folder (AppData).

hipersferus, 2021-07-21:
@rburema - I have the Cura printer config files (extruder, def and quality) for FLSUN SR printer (that is not officially added yet in Cura). I've copied the files added in the correct resources paths, and the profile is seen for Cura 4.9 to add the SR printer.
In Cura 4.10 it seems that the files are ignored, profile not seen. 

My workaround for this was to backup the profile in Cura 4.9, then restore it in Cura 4.10, and it worked. Printer is added and print profile works, even printed some parts and all is good from 4.10

It might be that FLSUN just needs to update the SR printer profile to be compatible with Cura 4.10

Side question: How does a new printer profile get added officially in Cura? It must be submitted (via GITHUB) by the printer producer only, just to make sure is a valid official profile?

Thanks!

Ghostkeeper, 2021-07-21:
We also accept profiles made by the community, not just by manufacturers.

rburema, 2021-07-22:
@hipersferus To come back to what @Ghostkeeper said earlier: 'The correct resource paths' are different for bundled printers and added printers. That's what he meant with the program files folder vs the appdata folder.

closed 2021-08-04 by no-response[bot]

nexteurope, 2021-08-31:
Hello
download cura 4.11 beta, the FLSUN Super Racer is available there.
https://github.com/Ultimaker/Cura/releases/tag/4.11-beta

rburema, 2021-09-01:
@nexteurope Careful though! There was a PR ([this](https://github.com/Ultimaker/Cura/pull/10192) one) added between the beta and the (eventual) release for that specific type of printer (FLSUN Super Racer) We're still deciding on whether to make a version-upgrade. If we don't, then please see the last comments in the linked PR.

Worst case, a reinstall may be recommendable.