Ultimaker/Cura #10140: Can't add material

LukasMueller187, 2021-07-15:
### Application Version

4.10

### Platform

Arch Linux

### Printer

Prusa Mk3s

### Reproduction steps

1. Fresh install, set up printer using the dialog.
2. In main window --> manage materials 
3. Trying to create new materials using the "+ create" button

### Actual results

Nothing happens. The list of materials is empty, none is added. The fields on the right are not editable besides the cost and the weight.
[image]

log:
`2021-07-15 12:00:05,989 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [421]: file:///usr/share/cura/resources/qml/Preferences/Materials/MaterialsDetailsPanel.qml:91: TypeError: Cannot read property 'container_node' of null
2021-07-15 12:01:44,905 - WARNING - [MainThread] cura.Machines.VariantNode.preferredMaterial [103]: Could not find preferred material generic_pla with diameter 2 for variant empty_variant, falling back to empty_material.
2021-07-15 12:01:44,930 - INFO - [MainThread] cura.Machines.Models.MaterialManagementModel.duplicateMaterial [198]: Duplicating material empty_material to custom_material #4
`

### Expected results

New material is added and editable

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[complete_log.log](https://github.com/Ultimaker/Cura/files/6822219/complete_log.log)

fvrmr, 2021-07-15:
Hi @LukasMueller187 do you use the Cura AppImage?

Ghostkeeper, 2021-07-22:
No, this was not with the AppImage. It looks like the AUR package is missing dependencies. This bug is missing because it's missing the fdm_materials repository. In the log it also appears like they're missing the pyzeroconf library.

We don't maintain the AUR package. It is up to their maintainers to ensure it works correctly, so we'll close this issue. We do provide our own AppImage to support most Linux environments: https://github.com/Ultimaker/Cura/releases/download/4.10.0/Ultimaker_Cura-4.10.0.AppImage

closed 2021-07-22 by Ghostkeeper