Ultimaker/Cura #10142: error en segmentacion

jptimber, 2021-07-15:
### Application Version

4.10

### Platform

catalina

### Printer

anycubic chiron

### Reproduction steps

La segmentacion del archivo se realiza mal en su base superior.   Se adjunto imagen donde se ve que el la base superior crea una impresion extraña y no lisa como debe ser.

### Actual results

error de segmentacion

### Expected results

la impresion debe ser lisa

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[500x120_inv-1.zip](https://github.com/Ultimaker/Cura/files/6824545/500x120_inv-1.zip)
[AC_500x120_inv-1.zip](https://github.com/Ultimaker/Cura/files/6824546/AC_500x120_inv-1.zip)
[image]

GregValiant, 2021-07-15:
I can't download the file.
In your screen shot - the area below the jagged line appears to be on a different layer.  I'm thinking that the floor of the model is not flat, or the model is not flat on the build plate in Cura.  You can see that the jagged line is made up of an outer-wall and backed up by 2 inner-walls.

Ghostkeeper, 2021-07-22:
Please write bug reports in English. I can't understand Spanish or Portuguese or whatever that is. So I'll pretend that you're an alien creature talking gibberish at me and giving me those pictures and that project file!

I've fixed the links in the original post.

The model has a difference in height there. It's pretty subtle (note that the units are um, not mm, due to Blender interpreting STL units as metres instead of millimetres).
[image]

So one side is 1 micron higher than the other.

Unfortunately, you have a layer height of 0.2mm, and initial layer height of 0.2mm too. That means that there'll be a layer spanning the Z coordinates 3.0mm - 3.2mm. With Slicing Tolerance set to Middle, it'll then slice at Z=3.1mm, exactly within the 1 micron difference in height there. That's why one side gets an extra layer and the other doesn't. It's the closest approximation of your surface that can be achieved with that layer height, technically.

Therefore this is expected behaviour. Not desired, but the model is at fault here, not Cura.

closed 2021-07-22 by Ghostkeeper