Ultimaker/Cura #10145: Z-hop on first layer only.....or to layer # = x

PhilBaz, 2021-07-15:
### Is your feature request related to a problem?

Not a bug, but a problem with printing first layers where there are lots of islands and the nozzle cates any little curl at the edge.

### Describe the solution you'd like

Z-hop on first layer only, or up to specified number of layers.

### Describe alternatives you've considered

tried different settings, but this would be a valuable feature for sure.

### Affected users and/or printers

Everyone will benefit if they have lots of small parts and trouble with first layer ....

fvrmr, 2021-07-16:
Hi @PhilBaz thank you for your feature request. I will discuss this with the team. Keep you posted!

PhilBaz, 2021-07-17:
Thank you! I think it will be a useful feature.

fvrmr, 2021-07-22:
Hi @PhilBaz I have discussed it, and it was not really clear to us why it would be an improvement. Do you see that it helps when you use z hops for whole your print?

PhilBaz, 2021-07-22:
Hello,

Yes, it minimizes the possibility for the nozzle to peel off another part
as it passes over. If there is any curling at the edge (which can happen)
the part will get peeled up/off without Z-hop.  And...I don't like to use
Z-hop because of increased stringing.

This might also be influenced by which combing setting you use. The
likelihood of an island being peeled up.

Ghostkeeper, 2021-07-30:
That's a good argumentation.

I'd expect that Avoid Printed Parts When Traveling already prevents most of the hits if you're printing a bunch of parts. However that can't prevent hitting the contour of the part you're printing the skin of.

We've added this to our backlog. (Developers, see issue CURA-8451.)