Ultimaker/Cura #10146: Cura Update 4.10 Ultimaker S3 problem

FabioGS3, 2021-07-16:
### Application Version

4.10

### Platform

Windows 10

### Printer

Ultimaker S3

### Reproduction steps

1. Updated Cura from 4.8.0 to 4.10.0

### Actual results

1. Update worked fine, Cura works besides lost Settings.

### Expected results

1. Problem: After slicing models and wanting to print them, a prompt on the Ultimaker S3 comes up telling me the used Cura Version is outdated, please use Version 4.4 or higher. Models were sliced with Cura 4.10.0. Screenshots attached of Cura
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

fieldOfView, 2021-07-16:
Check if there is a firmware update available on the S3.

fvrmr, 2021-07-16:
Duplicate of https://github.com/Ultimaker/Cura/issues/10062

closed 2021-07-16 by fvrmr

FabioGS3, 2021-07-16:
@fieldOfView  Worked, thanks !!!