Ultimaker/Cura #10147: no "skin overlap" when "top surface skin layers" enabled

MrStump, 2021-07-17:
### Application Version

4.10

### Platform

win 10

### Printer

CoreXY

### Reproduction steps

enable "top surface skin layers" and there is't "skin overlap" at any percent

### Actual results

no overlap

### Expected results

Expected that there will be "skin overlap"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

in 4.9 it did't work too, but in 4.9 it overlap skin layers **under** the "top skin", but don't in 4.10

fvrmr, 2021-07-19:
Hi @MrStump the skin overlap percentage is available for me even if the top surface skin layers are 0. Can you post a screenshot or share a project file, maybe there is something changed in your profile.

MrStump, 2021-07-19:
It's available, but didn't work, at all, it didn't overlap. I'll post project little later

Ghostkeeper, 2021-07-26:
We already have this issue in our backlog (CURA-6550). It is a duplicate of https://github.com/Ultimaker/Cura/issues/5758.

closed 2021-07-26 by Ghostkeeper