Ultimaker/Cura #10148: Crash when saving slices to media when using Bigtree3DPluginSuit

Hitamku, 2021-07-18:
### Application Version

4.10

### Platform

Windows  10

### Printer

Biqu B1

### Reproduction steps

Just do the install steps like the previous Cura

### Actual results

The result is the same, crash when saving the slacing result

### Expected results

On Cura 4.9 or earlier versions there is no problem. Only occurs on 4.10

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[DxDiag.txt](https://github.com/Ultimaker/Cura/files/6835512/DxDiag.txt)
[cura.log](https://github.com/Ultimaker/Cura/files/6835513/cura.log)

Hitamku, 2021-07-18:
[image]

konskarm, 2021-07-19:
Hi @Hitamku. We do not maintain this plugin, so please report this issue in the plugin's repository https://github.com/bigtreetech/Bigtree3DPluginSuit.

closed 2021-07-19 by konskarm

Hitamku, 2021-07-20:
Ok thank you