Ultimaker/Cura #10149: Stop Layer Separation (a.k.a. Delamination) with cross-layer stitches

eGit, 2021-07-18:
### Is your feature request related to a problem?

When Layers separate and printed parts break. They always break in Z-printing-direction for obvious reasons.

### Describe the solution you'd like

Little holes are left open in the layer below. The holes are filled while the second layer is printed, effectively stitching those two layers together. Layer 1 and 2 are cross-stitched this way. Also Layer 2 and 3, 3 and 4 and so on.

Since you fill the hole with a single "blob" of molten filament, it is like a nail that spans several layers. Similar techniques are used in certain types of wood-processing, where layers of thin wood are glued together and additionally fixed together over several layers with metal pins or glued in wooden pins. With the pins the finished results breaks around 200% of the strength the pin-less part endures.

The holes are always at different positions for layer pairs and not on top of each other, which would reduce the strength.

The holes could span more layers then 2. The maximum is probably 3 or 4.

### Describe alternatives you've considered

Real 3D printing would possibly also greatly improve the problem of layer separation, although it would be much harder to implement: https://www.youtube.com/watch?v=gmePlcU0TRw

### Affected users and/or printers

Everybody with a SLA printer

eGit, 2021-07-18:
Pins could be located only in the shell of the object. If this feature is active the shell could be printed automatically thicker at the pin-hole locations.

This would not only make the part stronger since it strengthens the shell, but also would greatly ease implementation of this new feature.

fvrmr, 2021-07-19:
Hi @eGit is your request only for SLA printers? 
The non-planar slicing is a duplicate of https://github.com/Ultimaker/Cura/issues/5980

eGit, 2021-07-19:
> Hi @eGit is your request only for SLA printers?
> The non-planar slicing is a duplicate of #5980

I only have experience with SLA printers, so I don't really know is this would make sense for other types of printers.

The non-planar slicing I mentioned in the "Describe alternatives you've considered" section would somewhat ease the problem of layer separation. But it would be only a side effect and non-planar slicing has not really anything to do with my feature request. I guess I should have left the "Describe alternatives you've considered" section empty.

I doubt the feature I asked for is related to any other issue. I tried my best to use the search function for something similar existing.

nallath, 2021-07-19:
Adding drawings really helps to make it easier to understand what your proposed method is (text & slicing strategies don't really play well together in my experience!).

One of the ideas that I had about this actually made it to a patent application: https://patents.google.com/patent/WO2020046125A1. It could very well be that your idea is already patented, which would make it impossible for us to implement it (unless of course it's the one that I came up with ;)).

nallath, 2021-07-20:
Ah yes. Unfortunately, that strategy is patented. I can't directly find the patent, but I've discussed it with some of my colleagues at some point.

whyme12, 2021-07-25:
@nallath If I understand your patent correctly, you are describing the fusion of two layers by "drilling" into them with the hotend. Interesting approach, but wouldn't this cause problems with the nozzle getting dirty and maybe even clogging.

@eGit:
Couldn't this also be done by printing:
- 3 perimeter walls
- both outer perimeters are printed normally
- inner perimeter is printed like a dotted line, with double layer height and alternating every layer to bond the layer below and leave a pocket for the layer above.

Here is an example, viewed from above:
[image]

Not sure if this is patented as well or maybe even falls under the same patent (hate those stupid patents holding 3d printing back...). But maybe its not, as no additional Z movement would be needed.

eGit, 2021-07-25:
@whyme12 in principle the effect would be similar. Parts of 2-or-more layers are printed (heated and cooled) at the same time. The little pins I suggested are bigger then a single line I would think and I would assume they would "nail" layers better together then what you suggested.

STILL. Your approach is also very good and if it is not covered by patents it should be even easier to implement.

Btw, I agree on patents. I mean if you really invent something groundbreaking. But what I suggested shouldn't be possible to patent in my opinion. The same concept is used in hundreds other areas to strengthen materials.

Ghostkeeper, 2021-07-26:
What would be the best way to "leave holes" in the lower layer, would you think? You could have the nozzle stop, retract, move 0.8mm, then unretract again and continue printing. But that's going to take a lot of time and wear down the material fast. It would also produce blobs, which by themselves aren't really a problem if it's on the inside but that material has to come from somewhere, so it means underextrusion elsewhere if there are a lot of them.

By its nature this could only work if lines are printed on top of each other in different layers. Unless the user is printing something completely vertical (which is hard to explain to the user) this will only happen in parts of the infill and skin areas. You couldn't realistically apply this technique to walls.

nallath, 2021-07-26:
> Btw, I agree on patents. I mean if you really invent something groundbreaking. But what I suggested shouldn't be possible to patent in my opinion. The same concept is used in hundreds other areas to strengthen materials.

Unfortunately, patent law disagrees with you. Incremental improvements are also patentable. Even if something is already done in another field, there are situations where you can patent the "old" method for the new field.

eGit, 2021-07-26:
> Unfortunately, patent law disagrees with you. Incremental improvements are also patentable. Even if something is already done in another field, there are situations where you can patent the "old" [...]

 Thanks. I know. Still doesn't mean I have to like it. Patents are counterproductive to any kind of progress as a whole.

eGit, 2021-08-11:
What if someone wrote a plugin with an anonymous account on Github? I am not talking about breaking any patents. I am talking about keeping the author of a free plugin safe from any harassment.

nallath, 2021-08-12:
> What if someone wrote a plugin with an anonymous account on Github? I am not talking about breaking any patents. I am talking about keeping the author of a free plugin safe from any harassment.

That could be done. But we would only be able to merge it if we checked if it didn't infringe on a patent, since we would accept responsibility in that case. 

The plugin structure of Cura is primarily focused on the frontend and since most of what needs to happen here is in the backend, I don't think that it's the best solution on a technical level either.

Anyhow, what does help is keep talking about solutions for this problem. The more ideas we put out in the world, the harder it will be to actually patent this (as all publicly available information is seen as prior art).

eGit, 2021-08-12:
Sure. Prior art. Doesn't keep people from filing patents and sometimes getting them approved. The patent process cannot find all prior art. And once a patent exists - even if it is bogus - Cura probably won't touch anything this bogus patent covers.

Can't people add plugins to Cura that are not contained in the official distribution? Hence plugins could be developed anonymously where people can add functionality covered by bogus patents and still have peace of mind because they know it is anonymous. Nobody wants to spend a lot of time for free developing such a feature and then deal with some patent trolls on top of it.

nallath, 2021-08-12:
> Sure. Prior art. Doesn't keep people from filing patents and sometimes getting them approved. The patent process cannot find all prior art. And once a patent exists - even if it is bogus - Cura probably won't touch anything this bogus patent covers.

Ah, but you can send prior art to patent offices even after it's approved. It's not a prefect situation, but it is possible. 

> Can't people add plugins to Cura that are not contained in the official distribution? Hence plugins could be developed anonymously where people can add functionality covered by bogus patents and [...]

That is possible, yes. Plugins can be "side loaded" as well. 

Anyhow, if someone makes a new algorithm that does something and provides that as a PR to us, we will do our due dilligence and check if it's patented. Once it's merged, it's essentially "our problem". So that would shield the original author (but do note; I am a lawyer in any sense, so don't take what I say about this for the full truth).

eGit, 2021-08-12:
Maybe Cura should explore possible solutions to prevent layer separation before work starts that is in vain? There are at least two approaches mentioned in this issue which could serve as a basis for due diligence.

Ghostkeeper, 2021-08-19:
One of the reasons why there is no help article in the Settings Guide about this problem yet is because it pretty much never happens with Ultimakers :( I've seen layer splitting with large single-walled prints (>15cm) but other than those we don't have a lot of experience with it at Ultimaker.

Some research has been done to try to prevent it, but I don't think Ultimaker wants to publish that until they know they have something that works either.

eGit, 2021-08-19:
Ultimakers are very good, I know ;)

But we are not talking about wall separation during printing. My prints are flawless. We are talking about wall separation of parts that are used for tasks where they encounter big forces. It is known that the amount of force a part can take before it catastrophically breaks is ~2-3 times smaller in layer direction. Technical solutions such as the one here proposed are supposed to increase the force parts can endure in layer direction.